XCSoar/XCSoar #305: Vario gauge units incorrect

thermalling, 2019-12-23:
When units are set to ft/min, vario gauge shows incorrect readings.
For example, when Vario InfoBox shows 600 ft/min, gauge arrow reads just over 3, and gauge digital Gross reads 99.9 ft/min.  Same applies for sink values.
[image]
This issue occurs in Version 7.0_preview14 git: f43980a3e3 (Aug 27 2019)
and in current 6.8.11 Kobo.
Thanks.
Neil.

lordfolken, 2020-05-08:
There are two bitmaps set as background for the vario infobox. To me it seems that the bitmap is not exchanged when you change the lift units at runtime. Once you restart XCSoar it displays correctly. Can you confirm?

thermalling, 2020-05-16:
Hi lordfolken.
Thanks very much for looking into this.
I have re-tested. Behaviour is identical after restarting XCSoar.

Test process:
1) Open XCSoar in SIM mode
2) Change units to m/s
3) Test by incrementing altitude using +10 button -> OK per screenshot
[image]
4) Restart XCSoar -> SIM mode
5) Change units to ft/min
6) Test by incrementing altitude using +10 button -> same bug as previously
7) Restart XCSoar -> SIM mode
8) Test by incrementing altitude using +10 button -> same bug as previously

A couple of notes:
- The same bug appears on the Kobo in flight, not SIM.
- The bitmaps are not the only issue; the number 99.9 ft/min displayed on the gauge should read 600 ft/min, to match the value in the InfoBox.
- Something for later: On m/s, the gauge bitmap is not quite accurate, showing just over the expected 600m/s.  This one is much less urgent.

Thanks again, and please let me know if I can help.

MaxKellermann, 2020-05-22:
The bitmap is only reloaded after restarting XCSoar.
And there are only two background images: one for m/s and one for knots. I have no idea who in their right mind would use knots for vertical speeds, but that's what XCSoar's original developers implemented.
So there's no image for ft/min, so XCSoar does "something".

closed 2021-07-21 by MaxKellermann