XCSoar/XCSoar #316: Info: weather: overlay: overlay not translated in the UI

mlep, 2019-12-31:
In XCSoar 7.0_preview14-Android.
Menu Info > Weather > Overlay: the items "Overlay" is not translated into other languages.
And there is no string "overlay" to be translated in https://hosted.weblate.org/search/xcsoar/translations/?q=overlay&checksum=&offset=

lordfolken, 2020-05-04:
Your right.
https://github.com/XCSoar/XCSoar/blob/9041dc620a76ce72906021463f42ce58f0353425/src/Dialogs/Weather/WeatherDialog.cpp#L90

and

https://github.com/XCSoar/XCSoar/blob/9041dc620a76ce72906021463f42ce58f0353425/src/Dialogs/Weather/WeatherDialog.cpp#L94

Define Overlay and overlay correctly but its missing from the pot file:
https://github.com/XCSoar/XCSoar/blob/master/po/xcsoar.pot

I guess the pot file should be updated.

MaxKellermann, 2021-09-04:
This code has been disabled meanwhile (because DWD has terminated the service for us), and thus doesn't need to be fixed.
The reason I didn't mark it as translatable was because I didn't want to put a burden on translators for an experimental feature which maybe disappears or changes significantly, to avoid wasting their time.

closed 2021-09-04 by MaxKellermann