XCSoar/XCSoar #317: Kobo Clara HD blank screen

aa11bb22cc33dd44, 2020-01-01:
When turned on, the start screen is blank (but when you press the screen in the right place, menu options such as FLY and SIM still work)

lordfolken, 2020-05-04:
Version of XCSoar? Platform?

MarkusBagge, 2020-05-24:
Can confirm this with v6.8.13. After pressing the power-(hard)button, screen becomes empty/white. Drag your finger from bottom to top brings up the FLY/SIM startup screen. Easy if you know. Not easy if you don't :)

cefn, 2020-07-22:
I can confirm this also with v6.8.13 and v6.8.15 on Kobo Clara HD. Blank screen on startup but successfully triggering any button or interaction which makes the screen re-render presents the normal landing screen correctly. All subsequent writes to the screen and general XCSoar features seem to work for this model. 

It's as though the landing screen completed a render before the display driver had fully loaded for the Clara, hence that 'write' never made it to the screen. 

Where should I try to add a brief sleep to delay rendering the System screen in the case of Clara HD model?

I note that in https://github.com/XCSoar/XCSoar/blob/719cedbf614b037c75081fa86f7be3735fb0daf4/src/Kobo/Model.cpp#L58-L66 there isn't an entry for the Clara HD model so it's impressive that it works out of the box except for this.

I haven't tried getting a shell on the Kobo to be able to e.g. interactively run operations like... https://github.com/XCSoar/XCSoar/blob/719cedbf614b037c75081fa86f7be3735fb0daf4/src/Kobo/Model.cpp#L83
...is that a sensible way forward to be able to investigate the filesystem for this model and line up the Clara HD specific values  and driver invocations correctly?

cefn, 2020-07-22:
Wondering if it's to do with https://github.com/XCSoar/XCSoar/blob/e7066d5a716ac8d19bc1b7dec3a58820260360f3/src/Screen/FB/TopCanvas.cpp#L314-L315 potentially as the HD models seem to need to wait for a buffer sync. So introducing explicit model detection support could resolve a few things for the Clara HD including getting the DPI right.

flightninja, 2020-08-07:
From a bit of testing it appears (at least for me) that even forcing frame_sync = true for Clara HD, and hence calling Wait(); still results in the same blank screen issue at start up.

I'm wondering (only a guess) if this could possibly be because the Wait(); function makes use of an older 'API call', and it looks like the newer kernel / chipset in the Clara may potentially require newer 'API call'?

From looking through the code, the current implementation of the Wait(); function:
https://github.com/XCSoar/XCSoar/blob/a405ea3ba2af3c4798abd11caf65a4cfb3ad2e8b/src/Screen/FB/TopCanvas.cpp#L346-L349

... it uses MXCFB_WAIT_FOR_UPDATE_COMPLETE, which is currently defined here:
https://github.com/XCSoar/XCSoar/blob/a405ea3ba2af3c4798abd11caf65a4cfb3ad2e8b/src/Screen/FB/mxcfb.h#L161-L167

... looking at the source code for the kernel for Clara as provided by Kobolabs (https://github.com/kobolabs/Kobo-Reader/blob/master/hw/imx6sll-clara/kernel.tar.bz2) there is a file in "kernel/include/uapi/linux/mxcfb.h" which seems to be more recent, and has updated definitions for many things, including for MXCFB_WAIT_FOR_UPDATE_COMPLETE (which itself seems to have multiple versions v1, v2, v3 for different chipsets, and some use a new struct mxcfb_update_marker_data).

@MaxKellermann do you think this issue could be resolved by implementing a newer mxcfb.h? I'm not a c++ dev, and hence I'm not sure if this is even the root cause of the issue? Furthermore I can imagine it being a challenge updating to newer mxcfb.h while maintaining backward compatibility with older kernels / models?

MaxKellermann, 2020-08-07:
Oh no, they changed the kernel APIs/ABIs for newer versions? That's what happens when corporations hack their own drivers without upstreaming them - a huge mess!
Maybe this solves the problem, but as you said, it will be complicated to maintain compatibility.
What we certainly never want is separate binaries for separate Kobo products.

flightninja, 2020-08-07:
It seems the mxcfb.h file has been updated incrementally over the years by Kobolabs (for example, new definitions / changes were added in the GloHD kernel too it seems, not just for the Clara kernel). In order to maintain future compatibility (and with newer models as they come out in future) without needing separate binaries, do you think it's possible to have one "super" mxcfb.h file which tracks the latest file from Kobolabs kernel, but also includes any definitions for older models as long as not conflicting); and then use model detection function to make the correct ioctl() calls for the relevant model in the TopCanvas.cpp file? Or is that too messy?

MaxKellermann, 2020-08-07:
Sure, sync our copy with the "official" one, and do different ioctls depending on the detected model. That's the way to go.

flightninja, 2020-08-08:
I'm not sure if this helps, but after lots of searching on GtiHub, I found that NiLuJe seems to have done something very similar to creating this (what they call "Frankensteined") mxcfb.h which covers all versions here:
https://github.com/NiLuJe/FBInk/blob/master/eink/mxcfb-kobo.h

It seems this FBInk library is then used in the KOReader software, and then they have different functions to do the different ioctl calls for different versions here:
https://github.com/koreader/koreader-base/blob/9e4ae531c28ff35f43828056e88c18a3e80ff6ea/ffi/framebuffer_mxcfb.lua#L120-L135

Exadios, 2020-12-13:
So there seems to be a lot of differences between this model and previous models that have been worked on.

The Kobo Reader supplied kernel is version 4.1.15. Interestingly, this kernel comes with CONFIG_USB_OTG=y so all the dd work done with previous kernels should not be necessary for this one. But things like CONFIG_USB_SERIAL is not set and other needed drivers likewise.

It should be possible to handle any context switching by unloading and loading kernel modules.

There have been significant changes in the supplied graphics and IEEE802.11 drivers. Probably others as well.

Initially, using a Crosstools-ng build environment with gcc v4.9.4, I was able to build the kernel using the original Kobo Reader supplied kernel config but could not boot the result. The last indication was from U Boot saying "Starting kernel" after which it disappeared, never to be seen again. The original config had all the CONFIG_SOC_IMX* set. Disabling all these other than CONFIG_SOC_IMX6* produces a boot-able kernel. BTW Kobo Reader have setup U Boot to boot a zImage - not a uImage as in the past.

Plainly, the kernel produced by the Kobo Reader development environment being boot-able but the Crosstools-ng not being boot-able indicates that there must be some significant difference between the two environments. Whether this difference is going to bite us on the butt at some later date is to be seen.

The enabling of all CONFIG_SOC_IMX* leads me to believe that Kobo Reader are hoping to deploy a "standard kernel" for all their models. So similar kernels may start appearing in other late models.

lordfolken, 2021-11-09:
@aa11bb22cc33dd44 is still an issue with more recent versions of xcsoar?

commit 3d1a59d1 (2021-12-03): Fix startup for Clara

This adds a 0.1s delay in rcS file for Kobo to allow touch screen
and display to initialize properly.  this was specifically tested
for Clara HD (N249) but other models have had similar problems and
those models may now work better.

Closes #611
Closes #317
not sure if #515 is fixed because I don't have that model of Kobo
closed 2021-12-03 by MaxKellermann