XCSoar/XCSoar #319: change frequency at ATR833

Blaubart, 2020-01-02:
I connected my ATR833 with my OpenVario. If I like to set active frequency, I have to click button five times. One click dosen't change frequency.

fb, 2020-01-03:
Are you using "Set Active Frequency" from the waypoint details or are you using a frequency infobox?

Blaubart, 2020-01-03:
Hi Fabian,

I used it from the waypoint details.

regards
Dirk

fb, 2020-01-17:
You did not mention what versions of XCSoar you tried.

Are there any ATR833 users out there having the same problem?

Blaubart, 2020-01-17:
Sorry!! I use Version V7.0_preview10

r0w4n, 2020-01-25:
I'm using xcsoar 6.8.12 on an android phone, using a ioio connected to an atr833s via a rs232 shifter. I'm using the atr833 driver. The frequency is selected after the first press for me.

lordfolken, 2020-05-04:
Is this still an issue with the current openvario image? If so please reopen.

Blaubart, 2020-05-05:
With the new image I can not change frequency. I setup NMEA device to 9600 baud.

Blaubart, 2020-08-15:
No ideas how to fix it?

lordfolken, 2020-08-20:
Turn on DEBUG on the device (in the device) menu. Do all the operations you expect to work, Copy over the XCSOAR and NMEA logs from logs/ and .xcsoar dir. and attach them to the ticket. 
if you have to click the button multiple times, something is clearly wrong. This can be the serial connection (as in physical connection) (A common ground is something that helps.)
https://www.funkeavionics.de/wp-content/uploads/2020/08/01.142.010.71e_ATR833S_OI-Rev1.08_200805_WEB-PRINT.pdf
4.7.3.2 Connector D-SUB DE-9 Female for Remote Control
The baud rate of 9600 seems correct.

Blaubart, 2020-08-20:
I make some test this evening and it works. It seems that connector was not OK.

Blaubart, 2020-12-01:
Today I tested again, and it didn't work. I'm pretty sure my connector is OK now. I have to click at least 3-5 times to change frequency.
To be sure connector is OK I like to send the string to change frequency manually via PC. Which string do I have to send?

Blaubart, 2020-12-05:
I tried to get a log file:

> Turn on DEBUG on the device (in the device) menu. Do all the operations you expect to work, Copy over the XCSOAR and NMEA logs from logs/ and .xcsoar dir. and attach them to the ticket.

in .xcsoar I only have one log file (xcsar.log) but it is a couple days old.

Blaubart, 2020-12-27:
Today I connected OV and the radio to my computer and looked to the data stream.
OV sends for example 02 72 12 84 03 E7 if I like so set standby frequency to 132.015 every time I click "Set standby frequency" . So it seems driver is working perfect. If I send with my computer 02 72 12 84 03 E7 to the radio standby frequency is set to 132.015, always!

I have no idea why it dosen't work, when OV sends frequency to the radio.

Blaubart, 2020-12-28:
It works! A friend gave me the correct hint. I connected GND from remote to BAT(-) instead of PIN 21 or any other internal GND.

closed 2020-12-28 by Blaubart

BenniBastler, 2023-04-27:
> I'm using xcsoar 6.8.12 on an android phone, using a ioio connected to an atr833s via a rs232 shifter. I'm using the atr833 driver. The frequency is selected after the first press for me.

Hello r0w4n,
that is exactly what I have been trying to do for half a year!
Since 2014 I have been using an IOIO with two RS232 inputs (ANDROPORT) and have connected FLARM and a Zander ZS1 to it without any problems.
Now I also wanted to connect the ATR833S to my Android tablet, so I got a Soartronic IOIO UART interface board V.2e, which has three serial RS232 inputs and one TTL input.
FLARM and Zander worked immediately without problems, but I can't get any data from the ATR833.
The f.u.n.k.e. support does not even know what an IOIO is!!

- What is the exact configuration of your devices?
Which HW version does your ATR833 have? (for me it is HW 1.0)
Which SW version does your ATR833 have? (for me it is SW 1.03)
Supposedly nothing has changed on the serial interface since then.
- Which IOIO do you use?
- Why did you use a shifter?
(as far as I know, it brings a TTL input to RS232 level).
But the serial interface of the ATR833S is already on RS232 level according to the manufacturer's support?

I would be very grateful for any advice,
If this had worked for me last year, I'm sure I could have avoided damage to the landing gear!
I have attached the documentation of my connection cable.

Greetings from Ben from Berlin
[image]

Scumi, 2023-04-28:
> but I can't get any data from the ATR833

Getting data from the ATR833 was just implemented recently (#1149). You either have to compile current master branch yourself or wait for 7.29.
Have you tried if setting a frequency works? That did before already and if it doesn't, then its probably your cabling.

BenniBastler, 2023-04-28:
Yes, I have connected my ATR833 with a Windows 10 PC directly via a normal serial interface.
The Frequency-Tool from f.u.n.k.e. was able to work with my ATR833 exactly as described in the manufacturer's manual,

BenniBastler, 2023-04-29:
Hello Scumi,
thank you for your interest in my problem.
No, it was not possible for me to get a connection via IOIO from ATR833 to XCSoar at all.
That is, the IOIO is of course connected to the Android device, but no data comes from the ATR833 as from the other devices.
Of course, I have also swapped TX and RX on a trial basis, but to no avail.

It's nice to see that Max Kellermann continues to improve the driver for the ATR833, but I haven't gotten to the point where it would make a difference yet.

Scumi, 2023-04-29:
The latest improvements on the ATR833 driver were made by myself so I'm curious what your issue is. Unfortunately your answers do not help me narrowing the issue down. It could be XCSoar, it could be IOIO, it could be something else.

- Which XCSoar version do you use?
- What EXACTLY do you do in XCSoar and what would be your expected outcome?
- How did you configure the device in XCSoar? Please post all configuration values.

BenniBastler, 2023-04-30:
Which XCSoar version do you use?
I have on my Google Nexus 7 tablet the last version 7.28 and on my Dell Streak 5 the last version that runs on it (Android 2.3.3), version 6.8.17.
My problem is certainly not the XCSoar version, because r0w4n commentet on Jan 25, 2020: 
"I'm using xcsoar 6.8.12 on an android phone, using a ioio connected to an atr833s via a rs232 shifter. I'm using the atr833 driver. The frequency is selected after the first press for me."

BenniBastler, 2023-04-30:
- What exactly do you do in XCSoar and what result do you expect?

I want to control the frequencies of the ATR833 via the moving map of XCSoar. Especially if I have to land at a foreign airfield, I would like to ask beforehand via radio, if there is not an invisible obstacle hidden somewhere, or if something else has to be considered.
My radio is somewhat inaccessibly hidden behind the control stick - adjusting the radio frequency by hand takes too long when you are about to make an outlanding but are still struggling with a weak updraft.

BenniBastler, 2023-04-30:
- How did you configure the device in XCSoar?
  Please provide all configuration values.

I have tried three different IOIO (Androport, SoarTronic, and a homemade one with IOIO OTG board), all work with my FLARM, Zander ZS1 and Condor flight simulator.
I had set different baudrates, but I think 9600 is correct?
As driver I have selected ATR833.
The NMEA inputs of the IOIO are always connected to XCSoar as long as 12 V supply voltage is applied to the IOIO.
But the ATR833 always says "no data", no matter which UART or baud rate I select.

My question is mainly if and why I need a RS232 shifter?
I have the SoarTronic IOIO UART interface board V.2e, which has three RS232 and one TTL input.
I have also asked SoarTronic this, but so far I have not received an answer to my mail.

BenniBastler, 2023-04-30:
I haven't tried the TTL input in my SoarTronic board yet; the board is still on the plane, so far from home.
Would that make any sense?
I assume that [r0w4n](https://github.com/r0w4n) only needed a RS232 shifter to get a 4th RS232 UART.
Hopefully he will get back to me soon.
Is there any other way to reach him directly?
Maybe [he](https://github.com/r0w4n) doesn't even receive the messages from this forum anymore?

Scumi, 2023-05-02:
Okay, I am fairly sure now that XCSoar is not the issue. 

> My problem is certainly not the XCSoar version, because r0w4n commentet on Jan 25, 2020:

That really depends on what you intend to do or to see. You said before...

> but no data comes from the ATR833

...because that can only happen from version 7.29 onwards as the code to do that has been added **after** 7.28 release.

> "I'm using xcsoar 6.8.12 on an android phone, using a ioio connected to an atr833s via a rs232 shifter. I'm using the atr833 driver. The frequency is selected after the first press for me."

That explains what you really want to do. You want to set a frequency via XCSoar and that worked since a few years, correct.

> I had set different baudrates, but I think 9600 is correct?

Yes, correct. I thought maybe you have misconfigured the baudrate (happened to me as well at the start).

> But the ATR833 always says "no data", no matter which UART or baud rate I select.

Again, it can't before 7.29. But setting a frequency should still work.

I can unfortunately not help with the IOIO-stuff because I have never worked with those. I just wanted to check if my understanding of your issues are correct and if the ATR833 might be the cause, which I'm 99% sure not that this is not the case.

BenniBastler, 2023-05-03:
> Maybe I was unclear: Does setting a frequency via XCSoar work?

Hello Scumi,
Now it works for me 😅
Thanks for your patience and advice.
I can set the frequency with XCSoar, although "no data" is displayed in the NMEA manager.
Otherwise my ATR833 f.u.n.k.e. device is connected to the IOIO just like the FLARM or my ZANDER ZS1.
- Both the ANDROPORT and the SoarTronic board already have RS232 splitters.
Everything is there what you need.
The only problem was, that it irritated me, that XCSoar always shows "no data" in the NMEA-Manager and therefore I didn't try, if it doesn't work anyway.
(A radio does not send so much data).

best regards from Ben

(And I am sorry that I had made so many words because of this; I have deleted them now, because this can only make confusion in the future).

Scumi, 2023-05-03:
All fine, good that it works. 🙂
7.29 was released today. Try this version and "radio" (instead of "no data") should be shown and the ATR833 also transfers data to XCSoar.

Happy flying.

BenniBastler, 2023-05-16:
Hello Scumi,
now also the port monitor shows that the ATR833 is connected as "radio".
Many thanks for your work.