XCSoar/XCSoar #321: KRT2: Changing radio frequency from infobox does not work

fb, 2020-01-03:
I'm not able to change the frequency using the frequency infoboxes.

Tested setup: KRT-2 connected via RS232 to a computer running TARGET=UNIX.

fb, 2020-01-04:
In the current master, reading the frequencies from the KRT does not work at all. Without this feature the infoboxes will be empty and you simple won't be able to offset the frequency. I worked on that a bit and made some modifications. A part of this is already in PR #320.

The other problem is that we are trying to offset the frequencies in steps of 1000 KHz and 5 KHz. As long as the infobox is not empty and no radio is attached, adjusting the value in small or large steps will work. However this will result in invalid channels like 118.095 MHz.

In reality the frequencies are 8.33 KHz from each other which are rounded to 5 KHz in maps, ATC,  … So the gap can be 5 KHz or 10 KHz.    

```
  Freq   Channel
----------------
118.0750 118.080
118.0833 118.085
118.0916 118.090
118.1000 118.100
```

Maybe the `RadioFrequency` class could be extended to take care of this.

roeles, 2020-01-05:
I will think about a good way to solve it.
I've ordered a VHF antenna in order to test this on my bench with a real KRT2.

roeles, 2020-01-11:
@fb I'd like to start working on this. First I'll take a look at responding to the frequency exchange button on the KRT2, then I'll look at what to do with the frequencies.

However, I'm puzzled on what code I should base my changes.
First I saw a bugfix where you send data out as a response to receiving the 'S' character. Now I see nullptr checks in the same PR.
Can you point out to me what I can base my code on please?

roeles, 2020-01-11:
I think it makes sense to make the RadioFrequency carry two uint8_t members:
1. MHz
2. Counts of 8.33KHz

This should allow for all 8.33KHz channels as well as the old 25KHz channels as they are 3 multiples of 8.33.
A KHz-getter that returns a float could take care of the conversion from the true 8.33KHz frequencies to the rounded frequencies (which are multiples of 5KHz).

fb, 2020-01-11:
Am 2020-01-11 09:35, schrieb Roel Baardman:
> @fb I'd like to start working on this. First I'll take a look at
> responding to the frequency exchange button on the KRT2, then I'll
> look at what to do with the frequencies.
>
> However, I'm puzzled on [...]

That's right, I changed the PR to include nothing but a fix the bug that 
caused the crash I was experiencing.

For improving KRT2 support, I think a good starting point is my PR and 
cherry-picking 
https://github.com/fb/XCSoar/commit/ba75c2dd3d2be7ab1fca77b16a8dc6c7854004a3 
to it.

MaxKellermann, 2020-01-12:
> I think it makes sense to make the RadioFrequency carry two uint8_t members:
> 
>     1. MHz
> 
>     2. Counts of 8.33KHz

Why not just the latter? Having two fields which need a formula to calculate the real value is just complex and fragile. And bloated.

roeles, 2020-01-12:
> Why not just the latter? Having two fields which need a formula to calculate the real value is just complex and fragile. And bloated.

You are correct. I incorrectly thought that 1000KHz (1MHz) was not an integer-multiple of 8.33 KHz. It is.
I'll just use the latter.

kxmm, 2020-02-14:
I have compiled the sources and I am able to control the KRT2 using the PC Serial Interface.
Using the TaskIT BLE232 (Bluetooth Rs232 LE Adapter ) I am not able to connect. Printing error "Fehler" in the KRT2 Device. 
Using an Serial Bluetooth Terminal I am able to Receive the "S".

I have tried to narrow it down, to ...

  if (!port->WaitConnected(env) || !OpenOnPort(dump_port, env)) {
      LogFormat(_T("DoOpen Check Dump"));

      if (!env.IsCancelled())
      ++n_failures;
....
But my C++ is a little bit rosty (20+ yeras ago).
Would appreciate your help.
Michael

PR #341 (2020-02-20): Fix mechanism to communicate frequency changes made on KRT2.
Closes #321 
Closes #322 

@fb
closed 2020-02-20 by MaxKellermann