XCSoar/XCSoar #322: KRT2: Radio infboxes do not reflect changes made on the KRT

fb, 2020-01-03:
The current status is:
1. change standby frequency on KRT -- frequencies in infoboxes do not change
2. swap frequencies on KRT -- frequencies in infoboxes do not change

With https://github.com/XCSoar/XCSoar/pull/320/commits/ba75c2dd3d2be7ab1fca77b16a8dc6c7854004a3, the standby frequency will be updated when changed on the KRT.
In addition, the `C` command *Exchange of frequencies* would have to be handled by the driver. 

To be honest, I'm not sure if this issue is worth the effort. I could live without frequency infoboxes, so my personal solution would be to just remove the infoboxes :smiling_imp:

Everything I need is already working:
- I have a waypoint file containing frequency details.
- I can change to a frequency from that file with XCSoar without needing to touch the radio
- I can see the momentary value of active/standby frequency on KRT display

@roeles what are your thoughts?

roeles, 2020-01-04:
@fb Thanks for reporting!
I really want this functionality, because my KRT2 is somewhat hidden in my panel. It's hard to read, hence the need to mirror the current values on Infoboxes.

I want this code to remain in XCSoar and will invest my time into fixing/maintaining it.

**Question:** In an email to me you mentioned that setting the frequency from the Infobox panel did not work (or not all the time). Is that still true?

fb, 2020-01-04:
Okay. I understand that this feature is helpful if you can't see the display otherwise.

Regarding the first question see #321. I will add some details.

I do not have any load (or antenna) attached at the moment. Also no PTT button. To be on the safe side you could use a dummy in place of an antenna.

fb, 2020-01-07:
I think `SetComputerSettings().radio.active_frequency` should be only used to keep track of what is the momentary frequency setting.
Those properties were introduced together with the infoboxes here: https://github.com/XCSoar/XCSoar/commit/3481aa44a886119df1cf4461e719b295b58ec6ac#diff-eb136b0530683cd9dd72c7d6721e8227

`ActionInterface::SetActiveFrequency()` should not update `radio`. Device drivers  will take care of that (using `devices->PutActiveFrequency()`) making it a read only property from the user perspective.

- Change frequency directly on the radio: Device driver will update info boxes
- Change frequency from infobox: Frequency will be sent to radio. When it was sucessful, device driver will trigger info box update.
- Change frequency from waypoint menu: same

When you press ↕, it will swap active and passive frequency and send `0x02 'C'` (just the command and not updated freqencies). So to be sure to not mix anything up you have to be carefully with altering `active_frequency` and `standby_frequency`.

Maybe this will come handy: [krt2.py](https://gist.github.com/fb/aa964a3d7b142a8257ca275136cadfd9)

roeles, 2020-01-08:
> Those properties were introduced together with the infoboxes here: 3481aa4
> #diff-eb136b0530683cd9dd72c7d6721e8227

Thanks for the info!
I just got my setup working with the KRT2 and I reproduced the issues. Will
look into this in the next few days.

Meanwhile let's first get your bug fix merged, since that fixes pre-infobox
functionality.

roeles, 2020-01-19:
I have looked at this issue, and I believe I have found the underlying issue for both this problem and the one described in #321.

I've thought to simply about the fact that radio frequencies are now updated from 2 sites: the KRT2 and the infobox.
I have not used a Valdity item to indicate when the last update was made. Thus, when you alter the frequency from the InfoBox, it quickly gets overwritten from the KRT2. 
I believe the NMEAInfo structure that's passed to Device Drivers is re-used and not always cleared. This effectively means that the KRT2 driver repeatedly communicates the last settings.

fb, 2020-01-19:
So what do you think about this? `settings.radio.*_frequency` will be only updated from a device driver. (see my comment above)

For a smoother UX the "desired frequency" after sending a command to the radio could still be shown in the UI, visually indicating (e.g. red) that this is not yet the *real* frequency of the radio. When we know the frequency change was successful the infbox will indicate that.
However, when setting the frequency fails (e.g. because there is no radio) the infobox will degrade and show "*---*" after a timeout (like 3 s).

roeles, 2020-01-19:
I think you misunderstood what goes wrong. This is what I believe what happens when you set the frequency from the UI:
1. ActionInterface::OffsetActiveFrequency is called, which calls ActionInterface::SetActiveInterface. This overwrites ComputerSettings.radio.*.
2. The next time the KRT2 is asked for new information, it returns the last information known to it (since the NMEAInfo structure appears to be re-used). This again overwrites ComputerSettings.radio.* via ActionInterface.

Thus, we should check if the KRT2 has **new** information and only then update ComputerSettings via ActionInterface.

fb, 2020-01-19:
I overlooked 0bca8ff9 (ApplyExternalSettings/RadioProcess()) which calls ActionInterface::SetActiveFrequency(). :astonished: 

My concern is if it will work reliably to use settings.radio.active_frequency like this:
(from xcsoar) desired frequency --> info.radio.active_frequency <-- actual frequency (from radio)

However, that seems to work for MacCready, Bugs, …; so it should work similarly for frequencies.

roeles, 2020-01-20:
I have modeled my code after maccready, since you can also offset it.
I think the developer manual has a picture of how information flows.
Perhaps that gives some insight on why/how it works.

kxmm, 2020-01-20:
One question: Which Hardware do you user to remote control the KRT2?
Regards Michael

roeles, 2020-01-21:
I soldered the wire ends of an Ftdi cable onto the plug of the KRT2.

fb, 2020-01-21:
@roeles can you test https://github.com/fb/XCSoar/commit/krt-swap-freq with your setup?

roeles, 2020-01-21:
@fb Sure thing.
I wonder if you're meant to modify settings directly from a driver though. My approach is to fill NMEAInfo, and then implement something in ApplyExternalSettings.. making it call ActionInterface.
@MaxKellermann does fb's solution look good to you?

fb, 2020-01-21:
We have to be careful not to mix things up. This issue was not about 
actions triggered by the user in XCSoar.

Let's look at it from the UX perspective:
You could change a frequency on the radio. Both frequencies in the UI 
should now be the same as on the radio. At least that's how I understand 
it's supposed to work.

Things get more complicated when the user also changes the frequencies 
from the UI (which could fail). But that's another question that we can 
discuss in #321.

fb, 2020-01-21:
> I wonder if you're meant to modify settings directly from a driver
> though.

Good point, the other drivers use setters from ExternalSettings

roeles, 2020-01-21:
I have successfully implemented setting frequencies from both UI and the
KRT2.
I just have to implement swapping in the same way and clean up, so perhaps
it makes sense to just wait a bit?

Roel

kxmm, 2020-01-21:
So it would be possible to use a Bluetooth 2 rs232 adapter. Like this one...

https://www.amazon.de/DSD-HC-05-Bluetooth-Pass-Through-Drahtlose-Kommunikation/dp/B01G9KSAF6/ref=pd_aw_sbs_147_1/260-6244147-9589327?_encoding=UTF8&pd_rd_i=B01G9KSAF6&pd_rd_r=b1c8812e-bae9-46f5-b730-152b3c399bec&pd_rd_w=qmsS5&pd_rd_wg=Wnsxx&pf_rd_p=fa5594f7-1f2f-4d20-a36b-dca4a2835930&pf_rd_r=WACXP7BAHHAJE5MYCP16&psc=1&refRID=WACXP7BAHHAJE5MYCP16
And configure it as an nmea device.
Right? 

Regards Michael

kxmm, 2020-01-28:
I have bought a BLE232 from taskit and configured it 9600,8n1. Unfortunatelly i am not able to connect to the KRT2 using xcsoar. 
Using putty I just receive "S".
Using the KRT2 Manager I am able to upload frequencies using a Bluetooth Serial Adapter.
Is there anyone who can help me. 
Pls write a pm to michael@fam-maurer.at

roeles, 2020-01-29:
@kxmm I'm sorry, but I'm currently too busy with the code and other things to help you.
Please be patient and wait for updates to XCSoar.

kxmm, 2020-01-29:
Ok.  Where can I find infos regarding commands that I can send to the krt2. I would like to try my hardware setup using pitty.
Regards Michael 

Dr.Michael Maurer
Sent from Mobile

PR #341 (2020-02-20): Fix mechanism to communicate frequency changes made on KRT2.
Closes #321 
Closes #322 

@fb
closed 2020-02-20 by MaxKellermann