XCSoar/XCSoar #334: MapGen does not work

sl0m0ZA, 2020-01-20:
I have been trying for almost a week to generate a map of Utah, USA. All I get is 

Error: Generation failed!

(http://mapgen.xcsoar.org/status?uuid=cdb4f47d75cf969c03ab3f4d0123ba0e6406f34b)

lordfolken, 2020-01-21:
I'm on it. Broken infrastructure behind mapgen.

lordfolken, 2020-01-21:
Please use the pregenerated map until then: 
http://download.xcsoar.org/maps/US_UTAH_R9N_HighRes.xcm

lordfolken, 2020-01-22:
- [ ] install new hardware
- [x] install os
- [x] install zfs / zfs share
- [x] configure bonding nic
- [] configure vlan
- [x] install cluster software (opennebula)
- [ ] configure storage in opennebula
- [ ] add dmz vlan to switch
- [ ] add vlan to opennebula
- [ ] reimport vm from backup
- [ ] reconfigure frontend firewall for vlan
- [ ] any mapping for firewall

lordfolken, 2020-01-27:
Currently processing 140 maps of backlog.

closed 2020-01-27 by lordfolken