XCSoar/XCSoar #336: InfoBox - Last Recently Used

kxmm, 2020-01-20:
Es wäre toll, wenn man eine Infobox LRU Wayponts hätte. Durch Tippen auf diese würde man dann eine Liste der letzten ausgewählten Waypoints sehen. Im Fluge wechselt man oft zwischen Waypoints hin und her. Das umschalten auf die LRU Waypoints ist aber etwas mühselig.

fb, 2020-01-24:
So having a shortcut to show the list with "Recently Used" filter directly applied would help?
  
[image]

kxmm, 2020-01-24:
Hi Fabian,

yes, because selecting the type “recently used” is not very fast.

When I fly together with friends I typically switch between some points (Airports, Mountains) to communicate my position and to check the distance to the points  the other report.  

Another Idea I had would be that the LRU points are being displayed right away in the Select a Waypoint Dialog. They disappear when you start to enter a Name.

 

Would be great to have such an feature.

 

 

Michael

 

 <https://user-images.githubusercontent.com/51237/73053410-f0d1a680-3e87-11ea-894d-903147a9cd66.png> 2020-01-24-090002_709x913_scrot

lordfolken, 2020-04-20:
Btw. for tracking teammates there is the teamcode function. Also we support online tracking of "friends" via skylines, even beyond the flarm range.

lordfolken, 2020-04-20:
There is also the watched waypoints feature. This constantly updates the listed waypoints with glide path calculation and distance.

kxmm, 2020-04-21:
HI,

the „Watched Waypoints“ are to static. I have already implemented the feature and submitted it.

Regards

Michael

lordfolken, 2020-05-04:
implemented by #372