XCSoar/XCSoar #350: XCSoar renders a black screen on Cubieboard 2

kedder, 2020-02-22:
Disclaimer: this is a bit esoteric issue and I think I have found a workaround. I'm looking for advice on how reasonable it is from someone who knows XCSoar inner workings well (@MaxKellermann?).

We at Openvario are trying to port the device to modern DRM/KMS graphics infrastructure (https://github.com/Openvario/meta-openvario/issues/2). The device is based on Cubieboard 2 (which is of ARM NEON architecture). When XCSoar is compiled with OpenGL support, it starts, responds to keyboard input, but renders a completely black screen.

I was able to narrow down the issue to the single compilation flag in `debug.mk`:

```make
 # IEEE/ISO semantics, for example it is not interested in "errno" or
 # the difference between -0 and +0.  This allows using non-conforming
 # vector units on some platforms, e.g. ARM NEON.
OPTIMIZE += -ffast-math
```

When compiled without this flag, XCSoar renders everything just fine. With this flag, even a simple "hello world" OpenGL triangle is not visible.

My question is: how much are we missing without this flag? How critical is it for normal XCSoar operation?

lordfolken, 2020-03-04:
Can you be specific as to the complier version?

kedder, 2020-03-04:
Sure. It is cross-compiled to ARM with gcc-8.3.0

```
$ arm-ovlinux-linux-gnueabi-g++ --version
arm-ovlinux-linux-gnueabi-g++ (GCC) 8.3.0
```

roeles, 2020-03-16:
I suggest to read:
https://gcc.gnu.org/wiki/FloatingPointMath

And see if replacing -ffast-math by one of the flags which -ffast-math covers helps.

kedder, 2020-04-01:
It turns out the issue is no longer reproducible after upgrading to mesa-20.0.0. This was probably an issue with the new lima driver, that was solved in new mesa. So, closing this.

closed 2020-04-01 by kedder