XCSoar/XCSoar #351: DisplayOrientation setting does not rotate the screen

kedder, 2020-02-22:
I intend to run XCSoar in bare KMS mode (i.e. not in X11 or Wayland window) and compiling it with `ENABLE_MESA_KMS=y` option. The display is portrait-oriented and I want to make XCSoar render the UI rotated 90 degrees on my screen. There is DisplayOrientation setting on xcsoar profile that seem to be supposed to do that, however it doesn't seem to do it correctly: the graphics seem to be rendered on portrait-oriented canvas, but then stretched on the landscape-oriented screen (see screenshot below).

The end goal is to run it in this mode on Openvario device (https://github.com/Openvario/meta-openvario/issues/2), but the issue doesn't seems to be specific to that device - I can reproduce it on my laptop as well.

To reproduce, I'm building the XCSoar on my laptop (Thinkpad T460 with Intel HD Graphics 520 GPU) with this command line:

```
make -j$(nproc) ENABLE_MESA_KMS=y
```

I have changed my `~/.xcsoar/default.prf` to include `DisplayOrientation="1"` line. To run XCSoar, I switch to linux console (Ctrl-Alt-F3) and start XCSoar normally: `output/UNIX/bin/xcsoar -fly`.

What I get as a result is on the photo below. I expect to see UI rotated 90 degrees instead, so it is usable if I orient display in portrait mode.

[image]

While looking into it, I've found this piece of code in `src/Screen/OpenGL/Init.cpp`:
```c
UnsignedPoint2D
OpenGL::SetupViewport(UnsignedPoint2D size)
{
  window_size = size;

  glViewport(0, 0, size.x, size.y);

#ifdef SOFTWARE_ROTATE_DISPLAY
  projection_matrix = glm::rotate(glm::mat4(1),
                                  (GLfloat)Angle::Degrees(OrientationToRotation(display_orientation)).Radians(),
                                  glm::vec3(0, 0, 1));
  OrientationSwap(size, display_orientation);
#endif
  projection_matrix = glm::ortho<float>(0, size.x, size.y, 0, -1, 1);
  UpdateShaderProjectionMatrix();

  viewport_size = size;

  return size;
}
```

Note, that `projection_matrix` is assigned to some rotating matrix inside `SOFTWARE_ROTATE_DISPLAY` (which is indeed defined) block, and then immediately reassigned to `glm::ortho` matrix after that block. First assignment seems to be effectively lost. Naively commenting out the second assignment does not result in the rotated display though, but corrupts it (I end up with white rectangle on black screen).

That said, are my expectations of how `DisplayOrientation` works correct (is it supposed to rotate the whole screen)? Does the code I mentioned looks like a bug? Maybe there is some other way to achieve the rotation?

kedder, 2020-02-22:
Interestingly, XCSoar-6.8.9b rotates the screen as I would expect. So there is probably a regression somewhere in 7.0 branch (forgot to mention, I was building XCSoar from master in the report above).

[image]

kedder, 2020-02-22:
I bisected the regression to 67a9a193fa319029dc95b4f4c22c5e9d38a9884f

```
commit 67a9a193fa319029dc95b4f4c22c5e9d38a9884f
Author: Max Kellermann <max.kellermann@gmail.com>
Date:   Tue Sep 18 08:21:52 2018 +0200

    Screen/OpenGL: hard-code the use of GLSL
    
    Drop the "classic" (shader-less) OpenGL code.  All of our targets are
    capable of GLSL, so there's no reason to keep the old code around.

```

roeles, 2020-03-16:
I've tried to reproduce your issue, since I'd like my recent KRT2 improvements to be available on OpenVario. (I have patches for xcsoar 7.0preview12 btw, in case you'd like to merge them).
The SOFTWARE_ROTATE_DISPLAY should be available on any OPENGL-platform. See src/Screen/OpenGL/Features.hpp:
`#if defined(ENABLE_OPENGL) && !defined(ANDROID) && !defined(TARGET_OS_IPHONE)
#define SOFTWARE_ROTATE_DISPLAY
#endif`

In turn the **ENABLE_OPENGL** flag is enabled in **build/opengl.mk** via **OPENGL**.
If I build a version with **OPENGL=y**, I can also reproduce the issue on X-Windows. 
I prefer that, since your build options produces a binary that screws up the nouveau drivers on my Dell XPS.

I reproduced the issue with the current HEAD (a397cf8d30).

roeles, 2020-03-16:
The issue appears solved when you replace:
`projection_matrix = glm::ortho<float>(0, size.x, size.y, 0, -1, 1);`
by
`projection_matrix *= glm::ortho<float>(0, size.x, size.y, 0, -1, 1);`

I think this makes sense, since you want to combine both the rotation and the orthogonal projection.

roeles, 2020-03-16:
I will try to test other orientations, since I don't know if **projection_matrix** is assigned anywhere else.

kedder, 2020-03-16:
@roeles, that's a great find! Simply replacing `=` with `*=` will probably not work in cases where `SOFTWARE_ROTATE_DISPLAY` is not defined, but that's trivial to fix! 

If you point me to the patch against current `master`, we can integrate it in openvario before it gets into mainline XCSoar.

roeles, 2020-03-16:
@kedder My patches have already been merged with mainline XCSoar. Nontheless, I can supply a patch against the current master.

roeles, 2020-03-16:
I have succesfully tested all 4 rotations on my machine.

kedder, 2020-03-16:
@roeles I mean patch for rotation fix. Do you intend to create a PR for XCSoar?

roeles, 2020-03-16:
Yes, as soon as I'm convinced that this could work, I'll create a patch for XCSoar. I'll link this issue.

roeles, 2020-03-16:
@kedder I believe my fix is correct, even when **SOFTWARE_ROTATE_DISPLAY** is not defined.
This is because **projection_matrix** is of type **glm::mat4** which is **mat4x4** via [1].
mat4x4 has a default constructor which initializes it as an identity matrix[2].
An identity matrix is essentially a null-rotation/translation. 
If you pre-multiply this with "something else", the effect is just that "something else".

[1] https://glm.g-truc.net/0.9.2/api/a00110_source.html#l00156
[2] https://github.com/g-truc/glm/blob/master/glm/detail/type_mat4x4.inl

roeles, 2020-03-16:
@kedder PR here: https://github.com/XCSoar/XCSoar/pull/363

kedder, 2020-03-16:
Thanks @roeles, great job! We can ship the latest XCSoar with all the great stuff with Openvario  now (including my #63, yay!) 

BTW, you can put "Fixes #351" into the PR description to make this issue automatically closed when PR is merged.

kedder, 2020-04-08:
Reopening. The fix (#363) was reverted because of rendering issues on Android. We'll have to reapply and verify the fix on android too. Revert commit is 6ef5768a073f7fd07442dc6f897f4fe7220d460e.

commit 8bd49afa (2020-03-16): OpenGL/Init: Fix software screen rotation

This solves a regression introduced in 67a9a19.
Correctly multiply the rotation matrix defined by software screen rotation with other transformations, instead of overwriting it.

Fixes #351
PR #373 (2020-05-14): Fix software display rotation (again)
Initial fix (8bd49afa25e112f3d5f377dfa1ad72656effc35d) caused rendering problems (black screen) on platforms in which `SOFTWARE_ROTATE_DISPLAY` is not defined (e.g. Android). It was reverted in 6ef5768a073f7fd07442dc6f897f4fe7220d460e.

Tested on Android, but was reproducible on regular UNIX as well. The problem with original `projection_matrix *= glm::ortho` fix is that `projection_matrix` is initialized with zero matrix.

Fixes #351.
commit 7317f965 (2020-05-14): Fix software display rotation

Initial fix (8bd49afa25e112f3d5f377dfa1ad72656effc35d) caused rendering
problems (black screen) on platforms in which SOFTWARE_ROTATE_DISPLAY
is not defined (e.g. Android). It was reverted in
6ef5768a073f7fd07442dc6f897f4fe7220d460e.

Fixes #351
closed 2020-05-14 by MaxKellermann