XCSoar/XCSoar #352: GPS Rollover problem on older android phones.

Mitkopeev, 2020-02-23:
Older android phones got affected by the GPS rollover second stage at the beginning of November.
The IGC files written by XC soar have the wrong UTC date stamp and are invalid for any claims.
Is it possible to get an internal fix in XC soar to correct the problem or maybe of the IGC file to be written with the phone own system time?

All the best.
Ivlelin

MaxKellermann, 2020-02-23:
On 2020/02/23 15:58, Mitkopeev <notifications@github.com> wrote:
> Older android phones got affected by the GPS rollover second stage at the beginning of November.
> The IGC files written by XC soar have the wrong UTC date stamp and are invalid for any claims.
> Is it [...]

XCSoar does not support broken/buggy hardware.

lyusupov, 2020-02-24:
@Mitkopeev 
LK folks seems to [had work this out already](https://github.com/LK8000/LK8000/releases/tag/7.1g).
If it is so - this could be a 'Plan B' for the hardware after that date.

closed 2020-02-24 by MaxKellermann

lordfolken, 2020-02-24:
They commited example code from u-blox. Commiting code with invalid liscences in the process (GPL,vs Public with Attribution) 
https://github.com/LK8000/LK8000/pull/1395/commits/043f22a9fc23fba063def6f4c2339842a4eb79de
The code basically assumes that a date of pre-20XX cannot be true. This is ugly and may cause all sorts of troubles.. a lot of devices start with a date 1970-01-01 when no timesource is configured.

Anyway. GPS hardware is cheap and abudent. Please buy working hardware or get a firmware update.