XCSoar/XCSoar #354: declaration to nano error message

memoryjb, 2020-02-24:
Samsung Tablet A running xcsoar v6.8.12 android, new nano latest update
 Continual error message, "task not uploaded"  any help would be appreciated

rawtaz, 2020-02-24:
@memoryjb You have to be much more specific and elaborate. Noone else knows what your setup/configuration is and how you are expecting the transfer to work. If you need help, provide relevant details, thanks!

memoryjb, 2020-02-25:
More information...
I am running xcsoar for gliding (realtime) and a lxnano flight recorder for verified traces (.igc files). What I require is that the task (a series or pre-arranged turn points) is achieved.  It is achieved by flying to each of the turnpoints.  The xcsoar is my navigation aid and the nano is the certified verifier (xcsoar is not a verified certifier).  Clearly both devices MUST have the same task in them.
This is a capability that both have, by way of once the task is defined in xcsoar as "task" xcsoar has a function (via bluetooth to the nano) to upload the task (its called a "declaration").  In that way both devices have exactly the same defined task.
I run xcsoar with an android device a Samsung Galaxytab A6. 
The xcsoar manual states to "declare task": Nav/Task/Manage/Declare. The Nav, Manage and task all work. But on selecting Declare, a dialogue box comes up with question "Declare task?" with Yes/no choice. When I select yes, the upload screen flashes for a second and then an error message comes up: "Error occured (with spelling mistake), Task NOT declared!"
If this can be fixed, I would appreciate it.

memoryjb, 2020-02-25:
problem solved.  In xcsoar :Config/devices/Lxnav/Edit/"passthrough device select "off" now it works flawlessly.

rawtaz, 2020-02-25:
@memoryjb Thanks for coming back with the solution so others can benefit from it :) It's usually about the settings, isn't it.

rawtaz, 2020-02-25:
@memoryjb BTW, please close this issue as the problem is resolved.

closed 2020-03-02 by memoryjb