XCSoar/XCSoar #355: Support OpenAIP airports, airspace... format

scls19fr, 2020-03-01:
Hello,

[openAIP](http://www.openaip.net/) is a web-based crowd-sourced aeronautical information platform which deliver free, current and precise navigational data to anyone.
Supporting their file formats could be a nice improvement.

See http://www.openaip.net/downloads for documentation
http://www.openaip.net/system/files/openAIP_aip_format_1_1_airport.pdf
http://www.openaip.net/system/files/Airspace.pdf
http://www.openaip.net/system/files/openAIP_aip_format_1_1_navaid.pdf
http://www.openaip.net/system/files/openAIP_aip_format_1_1_hotspot.pdf

Kind regards

lordfolken, 2020-05-04:
There is two issues with http://openaip.net
Any data that you contribute enters their copyright. You are contributing to their commercial product. 

The problem with openaip data is their license. https://creativecommons.org/licenses/by-nc-sa/3.0/ They do not allow commercial redistribution/usage which could be a problem for I.e. Stefly OpenVario, or even usage in a glider that is used for flight-school in a commercial setting.

Maybe you could negotiate an agreement with them? I'm for one quite sick of openair and cup files.

scls19fr, 2020-05-05:
> Any data that you contribute enters their copyright. You are contributing to their commercial product.

it's interresting to know but I don't think it's a problem for a XCSoar user who simply use their data without contributing

> They do not allow commercial redistribution/usage

Supporting this format in XCSoar doesn't mean that XCSoar should distribute this data.

> Maybe you could negotiate an agreement with them?

I'm not affiliate with them and they probably won't change their licence for a simple user.

MaxKellermann, 2020-05-05:
OpenAIP gets my full boycot for being a harmful project, luring people into believing they're doing something for the "community" when in reality they do unpaid work for a commercial firm. Writing this sentence is enough of my free time spent on that thing.

scls19fr, 2020-05-05:
Reading https://www.openaip.net/node/4

> Commercial licensing is currently not available. For more information on licensing, please contact us via contact@openaip.net

the commercial part is unclear to me.

Feel free to close this issue if you consider it not appropriate with XCSoar goals.

closed 2020-05-05 by lordfolken