XCSoar/XCSoar #356: xcsoar task with incorrect gps coordinates

memoryjb, 2020-03-02:
I am running xcsoar 6.8.12 for android on Samsung galaxytabA6.
I am using a "nano" lx brand flight recorder via bluetooth to source gps data in xcsoar, by connecting the devices in config/devices/B: LXNAV on bluetooth
I confirmed connection.
I declare a task that is already programmed on xcsoar into the nano. This works perfectly.
Problem: On flying the task, xcsoar indicates I have reached the turn point and indicates new heading to new turnpopint.  BUT, on finishing task, the .igc file shows I did not in fact reach the turn points.  Sorry cannot attached the .igc file, this site does not support attaching an .igc file.  This can be viewed on SeeYou app from Naviter and it show graphically that the turn points were not reached despite xcsoar notifying during flight I had reached the turnpopint.

memoryjb, 2020-03-02:
What I now know is that xcsoar task has turn points shown as 5km cylinder, but nano and xcsoar displayed on see you shows it as a sector.  Itr would seem as though there is a lack of data transferring?

MaxKellermann, 2021-08-23:
Without more data from you, this cannot be reproduced.

closed 2021-08-23 by MaxKellermann