XCSoar/XCSoar #357: further Bluetooth Low Energy support

UweRenschler, 2020-03-03:
We are currently releasing our Solario+ Variometer, which is a solar powered Bluetooth Low Energy based variometer, GPS, IAS...
Are there plans to support f.e. a VSP (virtual serial port) on Bluetooth Low Energy ? Currently our plan is to write an Android app, which translates from VSP (not SPP on Bluetooth 4.0) to a TCP port. For USB connection this is a good working solution, but generally we want to have BLE (no wires). Makes it more sense, if we contribute here into a driver or is it a better idea to do our own VSP to TCP/IP port android translator ? It would be great to get some feedback ( uwe@renschler.de ) from the developers.

lordfolken, 2020-05-04:
Please contribute to a driver. That would help.

UweRenschler, 2020-05-05:
Hallo Philipp,

 

Vielen Dank für die Info.

Gibt es zu XCSoar Infos wie man einen Driver in das Ganze einzubinden hat

ODER

Ist der Code Dokumentation genug um Driver zu schreiben ?

 

Ich würde mich über ein paar Worte sehr freuen.

 

Grüßle

Uwe

 

P.S.: Ein aktueller Bluetooth Gatt Server von Android läuft timing mässig sauber mit einem Service mit einer characteristic. In dem Moment wo man 4 characteristics mit unterschiedlichen flexiblen Latenzzeiten bedient rennt Android in diverse Betriebssystem Timing Probleme. Das sieht man auch an den Android Apps (FlyMe) die kein Notify können. Was vernünftiges baut auf die Nordic Lib auf, die auch sauber 4 Characteristics mit Notify kann, aber hier fragt es sich, wie man die Nodric Lib in das Gesamtkonzept von XCSoar einbindet.

 

P.P.S.: Ich dachte das ganze (XCSoar) ist Tod, da sich auf github kein Mensch meldet. (bei LK8000 ist das anders)

lordfolken, 2020-05-05:
Our Development Manual: (these are nightly builds, you may have to browse in the nightly_build paths)
https://download.xcsoar.org/nightly_builds/7.0_preview15~git%239d776759/manual/XCSoar-developer-manual.pdf

For a development environment you can use either the docker container or the vagrant box if you are not running debian10 as your main environment:
https://github.com/XCSoar/XCSoar/tree/master/ide

The procedure is also explained in the latest development manual.

Beware that your solution should work on a lot of platforms which are not android. (WINPC, Kobo, Plain Linux... ) So code platform neutral as much as possible

For drivers I'd recommend to look at the current drivers,:
https://github.com/XCSoar/XCSoar/tree/master/src/Device/Driver

Writing code that runs is quickly done. Writing maintainable code is 3 times as hard. That is why the quality standards are high and the code must fit into the rest of the platform. Also you should be willing to help maintain the code long term.

UweRenschler, 2020-05-06:
Hello Phillipp,

 

Thanks a lot. We mainly want to do an Android version and not more.

So I think ower own translator from BLE to a supported socket using the Nordic lib (for Android) is best.

 

Cheers

Uwe

 

https://download.xcsoar.org/nightly_builds/7.0_preview15~git%239d776759/manual/XCSoar-developer-manual.pdf

https://github.com/XCSoar/XCSoar/tree/master/ide

https://github.com/XCSoar/XCSoar/tree/master/src/Device/Driver

kedder, 2020-05-06:
Isn't BLE already supported? I used to connect to Chineese JDY-08 BLE couple of years ago on android and have seen the NMEA stream in XCSoar device. Don't remember doing anything special to support that.

UweRenschler, 2020-05-06:
No. Only BLE with SPP (serial port protocol). We use the same protocol like the SENSBOX from Flytec/Bräuniger and this is not supported.

Have you ever connected a SensBox to XCSoar ? (No NMEA stream, but a BLE Gatt Server with 4 Services and several characteristics.)

kedder, 2020-05-06:
> We use the same protocol like the SENSBOX from Flytec/Bräuniger and this is not supported.

I see, makes sense.

> Have you ever connected a SensBox to XCSoar ? 

No, I don't know much about SensBox and have never seen it.

MaxKellermann, 2020-05-06:
This feature request does make sense. XCSoar's BLE support is currently limited, and adding support for other BLE products is useful, so I'm going to reopen this.
But the problem is, as always: who wants to spend time on writing the code? Ideally, the vendor writes code to support his own product and submits it to XCSoar for code review and merging. If the code is well done, it will be merged. But the path to good code is long and difficult (for everybody of course, not just for vendors).
There is no documentation for writing a completely new way of attaching devices. Writing such extensive documentation would not be worthwile, because the time would better be spent on writing code. So anybody who wants to hack XCSoar needs to read existing code, and/or talk to people who know the code already.
Code using Android APIs needs to be written in Java, but XCSoar is C++, so there needs to be a bridge between Java and C++. That's not easy (nothing is).
Here's the Java part for talking to currently supported BLE devices: https://github.com/XCSoar/XCSoar/blob/master/android/src/BluetoothGattClientPort.java (written by @felixhaedicke)
This code implements the interface `AndroidPort`, and this class already has C++ bindings because it is shared with IOIO and "classic" Bluetooth.
If you implement something which doesn't work like `AndroidPort`, it needs a new (Java) interface with new Java/C++ bindings.

UweRenschler, 2020-05-06:
Thanks a lot for the link.

I will have a closer look at it and check if I see an easy way to use the Nordic Lib to interface to an AndroidPort.

I think I first will finalise our own App with BLE and the lib and see all timing related stuff. With this knowledge and the links below I will check the effort. 

I also can program C and C++ (mainly on Cortex-M Cpus for realtime software). Hints how Bluetooth is adopted in C++ on Android are welcome.

MaxKellermann, 2022-08-24:
Support for Bluetooth LE sensors has been added last year in version 7.11; this release contains a framework which makes writing new BLE drivers easy. I suppose this feature request is complete.
To support more sensors, a small device-specific piece of code needs to be written, but that's out of scope here. If anybody needs help with writing such a driver, contact me.

closed 2022-08-24 by MaxKellermann

iltis42, 2023-02-19:
I have added BLE advertising meanwhile in XCVario, follwing this setup here: https://openlabpro.com/guide/ble-uart-on-esp32-controller/ .
In principle in a terminal this works, i can see the data coming in as expected (1), hence XCS won't accept the device, configures a "BLE Sensor" with no option to configure a Driver protocol and stops talking to the device after a couple of seconds telling the device is unsupported.
Apparently to me this looks like my device cannot be configured properly in XCS, would the BLE experts give me a hint what we needed to do in order to get the device properly in when advertising BLE ?

1)
[image]
2)
[image]

1)
[image]