XCSoar/XCSoar #358: XCSoar beta - Android

scls19fr, 2020-03-06:
Hello,

Is there a simple method to install latest beta of XCSoar (without compiling it).
I'm not looking for a Play Store distribution but a .pkg file could be fine.
Ideally this kind of distribution should use an other configuration file (or an other folder) than the "production" ready version.

Kind regards

lordfolken, 2020-03-12:
There is the xcsoar-testing version on the play store.

There is a also a preview version here: http://download.xcsoar.org/nightly_builds/7.0_preview14~git%23923cf10d/ANDROID/

Its also already 3 months old.. so ymmv. 

All of these use the same datadir per default. The linux version allows you to specifiy another datadir:
```bash
-> % ./output/UNIX/bin/xcsoar --help
Free storage 2945160 KB
Starting XCSoar XCSoar v7.0_preview14-Linux~git#ecaf5aba53
Usage: ./output/UNIX/bin/xcsoar 
  -datapath=      path to XCSoar data can be defined
```

scls19fr, 2020-03-12:
XCSoar-testing for Android on Play Store is from April 2019 according
https://play.google.com/store/apps/details?id=org.xcsoar.testing&hl=fr

I wonder if you couldn't use Github Action to automatically create a beta release when tagging a new release on Github.

elgandoz, 2020-03-13:
There is a pull request open for this but has never been merged: https://github.com/XCSoar/XCSoar/pull/253

MaxKellermann, 2020-03-13:
> There is a pull request open for this but has never been merged: #253

Because it's marked "work in progress" [WIP], i.e. not ready for merging.

elgandoz, 2020-03-22:
The aforementioned PR has been now merged, without further changes. You can now find nightly builds at http://download.xcsoar.org/.

For Android specifically you can install the apk from `/nightly_builds/{version}/ANDROID/bin/XCSoar-debug.apk`.

Despite the latest version does work on my phone, this is great news, especially to check the most up to date manual.

bobmorane06, 2020-04-17:
that's very cool. ++

closed 2020-04-17 by lordfolken

elgandoz, 2020-08-18:
The nightly builds apparently have stopped uploading since 03/07/2020.