XCSoar/XCSoar #360: XCSoar 7 Android vs. Singleboard pc

GLDRLS, 2020-03-07:
Hi all,

I now use xcsoar7 with a raspberry pi 3b + and touchscreen

In xcsoar7 I missed the option to change the text entry.
The Android version is still providing this feature.
Is it a bug for Linux users?

Also and much more important, When I tap I to the landscape the features "map elements on this location" did not open except I touch in to a airspace.
But airfields or other waypoints are not shown.
This feature also in Android works fine.

I tried the same on stefly's openvario with touchscreen and "his" software
But the problems are the same.

All xcsoar versions are 7.0 preview 14

CazYokoyama, 2020-03-07:
What do you mean "change the text entry"? On menu -> config -> system -> setup -> weather -> pc_met_username, we have keyboard screen. Don't you have it?
I have it. My XCSoar is self built on rpi4b+ and touchscreen from master branch as of today,  c300c375115e8c4821c0e9932a255c5e6e57f535.

GLDRLS, 2020-03-08:
[image]

The upper one is in use but the lower one I want to use. 
Normally it's easy to change 
Config - > look - > language, input

In xcsoar7 (singleboard version) are only 4 points to choose 
[image]
This picture shows an openvario with his "official" software

[image]

This picture shows the same side on Android 

But much important would be this side
[image]
He is not proper working on my raspberry /openvario

CazYokoyama, 2020-03-08:
Sorry, I don't find the screens above in my xcsoar.

mihu-ov, 2020-03-10:
> I tried the same on stefly's openvario with touchscreen and "his" software
> But the problems are the same.
> 
> All xcsoar versions are 7.0 preview 14

Can you point me to "his" software? To my knowledge there is no Openvario image with XCSoar 7.0-preview14.

GLDRLS, 2020-03-10:
Yes,  it's my fault
Stefly - >
Version: xcsoar7_preview10
Date : Apr 15 2017
Git : 4e2485d

MaxKellermann, 2020-05-25:
So this bug report is about a version that is 3 years old?

GLDRLS, 2020-05-26:
No,
Maybe I wrote it confusing, I'm using the version 7.0 preview 14 git#72ba7ac89f

MaxKellermann, 2020-05-26:
So that's 4 months old.

GLDRLS, 2020-05-26:
Is this good or bad?
And which libinput library or source do you recommend?

MaxKellermann, 2021-03-18:
Old versions are bad.
What does `lsinput` say?

GLDRLS, 2021-03-28:
ouput lsinput

/dev/input/event0

bustype  :  BUS_USB
vendor    :  0x461
product   :  0x22
version    :  256
name      : "NHD Newhaven Display "
phys        :  "usb-3f980000.usb-1.1.2/input0"
uniq         : ""
bits ev      : EV_SYN EV_KEY EV_AB EV_MSC

closed 2021-07-08 by MaxKellermann