XCSoar/XCSoar #365: GPS fix interruption on HiSense A2 Pro Android 7.1.2

olivermaas27, 2020-03-17:
I installed XCSoar 6.8.9 on a HiSense A2 Pro (Smartphone with E-Ink screen on the back side) running on Android 7.1.2. Location determination via GPS works but is irregularly interrupted: Every 2-5 s "GPS waiting for fix" is displayed for 1-2 s.
I also tried XCSoar 6.7.9, but the result is the same.
I installed "GPS Test" - App, and there GPS fix is not interrupted. Also for other Apps like Maps.me the GPS input works fine.
Does anybody has an idea?

lordfolken, 2020-03-25:
most likely power saving option or wrong permissions. Also use 6.8.11/12. Maps.me and other GPS tools usually use location libraries, which correlate a variety of sources for position fix, such as wifi.

lordfolken, 2020-05-04:
No reaction and no more information. Closing for now.

closed 2020-05-04 by lordfolken

polmak65, 2020-08-10:
Hi, I also bought a Hisense A2 pro to replace my Yotaphone 2. It is a disappointment as I have the same issue with my GPS (makes a GPS fix every 2 to 5 seconds). I use Mach 7 to fly planes and it seems to work fine. Any ideas of how to fix this?