XCSoar/XCSoar #366: xcsoar 7pre14 on Galaxy Note3-Android5

lele70, 2020-03-17:
Hi,
I tried the beta on my Note3 where 6.8.12 runs with no issues but the app only opens with no infoboxes, maps, ...when you try to touch the screen it becomes full red. thanks

kxmm, 2020-03-31:
Hi,
I have the same issue. I thought it may be connected with the old android version so i installed Linage 16 with android 9. Still the same issues. No Info Boxes and when you tab on the screen it gets red.
Even when you start the app the Buttons Fly and Sim are not shown.

fb, 2020-04-01:
This is maybe the same problem I had with a XT1068 (Motorola G2) running Android 6.0. OpenGL renderer: `Adreno (TM) 305`.

@kxmm, @lele70 does it look like this on your devices?
[image]

I tested different versions to find the problem. The last working version is preview11. I didn't use `git bisect` but apks from a random Play Store archive site.
I can only guess that it could be caused by a change between preview11 and preview12: https://github.com/XCSoar/XCSoar/compare/v7.0_preview11..v7.0_preview12

[image]&nbsp;[image]

Looks like bitmap/texture rendering is failing, so I guess it's related to OpenGL.

kxmm, 2020-04-01:
Hello Fabian,

yes it looks like this (more or less – sometimes it shows something..). Is there any way to help you? 

I have downloaded the source and I am able to compile xcSoar….

 

Michael

 

 <https://user-images.githubusercontent.com/51237/72684281-7c8db080-3adf-11ea-80ed-b4af3d17fa21.png> 

 <https://user-images.githubusercontent.com/51237/72684286-8c0cf980-3adf-11ea-8203-b269dfb758e2.png>   <https://user-images.githubusercontent.com/51237/72684285-8a433600-3adf-11ea-8e09-0d534de57339.png>

fb, 2020-04-02:
Maybe you could also test if the rendering issues disappear with preview11. Then we can be more confident that it's the same problem.

kedder, 2020-04-02:
I might be confusing this issue with some other one, but I once bisected very similar looking issue (graphics defects on old android device) and bisect pointed to a commit that switched android SDK to higher version.

lele70, 2020-04-02:
The screenshots above show the same I see on my Note3 Android5 Adreno330 (Open GL ES3.0 V@84.0 AU@05.00.00.002.025) 
The app didn't crash, simply the map and infoboxes  are not shown. You can access the config menus and they works. When you switch from one page to the other the screen flash for a shot full red.
If I could have the APK of preview11 I will try to uninstall the preview14 and try it.

kxmm, 2020-04-02:
Does anyone know which APK was used for preview 11?

Dr.Michael Maurer
Sent from Mobile

elgandoz, 2020-04-03:
I have a different issue with the very latest previews builds. I'm getting this warning in the logs, repeated several times: Unsupported ChipID value for QGPU features (0x5030004), using a5x as default. I'm not sure if it is actually related. This happens on a OnePlus 3T running a custom Android 10 ROM. The last preview build I can run was the 7.0_preview14 from September 2019 (or so, it is now no longer available). XCSoar Test from play store works fine.

fb, 2020-04-11:
> I might be confusing this issue with some other one, but I once bisected very similar looking issue (graphics defects on old android device) and bisect pointed to a commit that switched android SDK to higher version.

I did a bisect and bff32731 (raise API level to 21) introduces the rendering issues, so it could be the same.

groundhogxc, 2021-03-03:
This is a duplicate of both #386 and #466. And it is a pity because there are apparently many very interesting devices affected.

I can also reproduce it, and it can be narrowed down to that clearly the issue is with displaying bitmaps. It is quite simply that everywhere where there should be a bitmap (texture) -- including buffered infoboxes -- that area shows up in a solid color (usually black or grey). I think the particular color is simply the last brush color used within that OpenGL context, that is why it is not always the same. By disabling some of the buffering code the infoboxes can be somewhat rescued, because the non-buffered boxes do show up. But of course many things do not work at all without bitmap rendering, so this does not really solve the issue.

All bitmaps are affected, both the ones loaded directly, and the ones rendered internally, so it seems it is not the creation of the bitmaps that is the issue. One thing I was wondering is whether it is the rendering itself or the "stenciling" that is actually failing here.

Max and @lordfolken said a developer needs such a device to fix it. I can't easily give my device away, but I do have one, I know basic C++ and I can compile the source. So I am open to suggestions as to what to attempt, since I have only very very basic OpenGL knowledge myself. Maybe someone has an idea of what to play with?

I did mess around a bit with the OpenGL precision flag settings, because that was mentioned as a solution for other people in an issue affecting the same GPU, but that did not help unfortunately.

MaxKellermann, 2021-03-15:
The bisect is interesting, and will be very helpful once I have such a device, but until that happens, I can't do anything.

lele70, 2021-03-16:
Hi MaxKellermann, if you have a device (Galaxy Note3), can you troubleshoot the problem? How long will you need to keep it? thanks

MaxKellermann, 2021-03-16:
A few days.

groundhogxc, 2021-03-25:
This bug has now become much more urgent, since it seems like @MaxKellermann has now released the 7.0 version. I think up to now people with this GPU could just stay with the regular release version, which was still working, and just not update to 7preview versions. Now all of those devices will stop working on the next update.

If someone has an idea of what I could try on my phone, let me know. Someone mentioned in a duplicate issue that the querying of GL capability properties could be insufficient, but I am not sure which direction to probe in that regard.

lordfolken, 2021-03-25:
@groundhogxc so sponser a phone for @MaxKellermann They cost less than 75Usd now.

rawtaz, 2021-03-26:
@groundhogxc I'll chip in $10 if you go for it (I'll Paypal you if you have Paypal).

lele70, 2021-04-01:
Hi guys,
just to say that I found on Googleplay an update of xcsoar and instinctively I clicked on it :)) on my Galaxy Note3. I then lounched the app and... It works!!! (head line says 7.0 but in the upper left corner says 7.1) As the icon was blue I didn't think it could be the final release of 7.0 testing that doesn't work with the Note3. What to say: I'm happy of it. Thanks Max.

MaxKellermann, 2021-04-02:
Interesting - I did not attempt to fix it yet (because I had no information what was wrong and had no device to reproduce it), but this change may have solved the problem: https://github.com/XCSoar/XCSoar/commit/2451e25e55d92a19c8c1081daf07641624548fde
Maybe some drivers don't allow `glLinkProgram()` to be called twice.

closed 2021-04-02 by MaxKellermann

lele70, 2021-04-02:
If someone else with Note3 or other devices that shows the same issue can confirm that official 7.1 is working, it could be nice.

fb, 2021-04-10:
On the device I created the screenshots with (XT1068 (Motorola G2), Adreno (TM) 305), in 7.4 the rendering issues are gone. :+1: