XCSoar/XCSoar #374: Fatal Error AllocatedString.hxx

kxmm, 2020-04-12:
Hi Max,
currently I am not able to build when I fetch the latest code from Github.
./src/Util/AllocatedString.hxx:37:23: fatal error: string_view: No such file or directory.
Would be great if you could fix this.
Regards
Michael

MaxKellermann, 2020-04-12:
This means your compiler is too old and doesn't support C++17.

closed 2020-04-12 by MaxKellermann

kxmm, 2020-04-12:
I am using the provided Vagrant environment.

MaxKellermann, 2020-04-13:
Then this environment is broken, but I have no idea of Vagrant (or even what Vagrant is).

kxmm, 2020-04-13:
For all vagrant User: Till the new vagrant environment "Debian 10"is available you have to upgrade to Debian 10:
change /etc/apt/sources.list:
  deb http://deb.debian.org/debian buster main
  deb http://deb.debian.org/debian buster-updates main
then upgrade using apt

finally gcc --version should show:
gcc (Debian 8.3.0-6) 8.3.0

than you can build the latest code from github without errors :-)

lordfolken, 2020-04-13:
#296