XCSoar/XCSoar #377: ANDROID: Permissions not in one go v6.8.13

lordfolken, 2020-04-12:
Hallo, 

Concering @v6.8.13
Observed on Android 7 and Android 8 devices:

If you install XCSoar.apk (6.8.13b) from the website, upon first launch XCSoar asks correctly for storage permissions. Once granted the dialog dissapears and you are returned to your last applicattion. XCSoar needs then be accessed from the notifications, after which it asks for location permissions. Once granted, you have to yet again access it from notifications and then it runs fine.

While not breaking, its highly confusing.

dernatsch, 2020-05-09:
I decided to look into this a bit and noticed where this behavior is caused.
When the permission dialog is opened the Method onPause() of the XCSoar class is called. A few calls follow that end up [here](https://github.com/XCSoar/XCSoar/blob/master/src/Android/Main.cpp#L289-L294), where `exit(0)` causes the application to close, apparently because the event subsystem is not initialized.

lordfolken, 2020-05-10:
Thanks for looking into this. That what you describe may be an additional problem.

As for the permissions:
So you can call for all the permissions at once described here:
https://stackoverflow.com/questions/40169809/how-to-ask-and-accept-all-permissions-in-only-one-confirm-alert-dialog-in-androi

The requestPermissions function can take an array of permissions:
https://developer.android.com/reference/android/support/v4/app/ActivityCompat#requestPermissions(android.app.Activity
```java
int PERMISSION_ALL = 1; 
String[] PERMISSIONS = {Manifest.permission.READ_CONTACTS, Manifest.permission.WRITE_CONTACTS, Manifest.permission.SEND_SMS, Manifest.permission.ACCESS_FINE_LOCATION};

if(!hasPermissions(this, PERMISSIONS)){
    ActivityCompat.requestPermissions(this, PERMISSIONS, PERMISSION_ALL);
}
```
We currently request one after the other here in the requestAllPermissions function defined here:
https://github.com/XCSoar/XCSoar/blob/bc1dab4e72a0550f12772a46ff6db7d84ab5f0a2/android/src/XCSoar.java#L236

dernatsch, 2020-05-10:
@lordfolken Thanks for the input!
I opened PR #392

lordfolken, 2020-05-14:
Closing this as its fixed in code.

closed 2020-05-14 by lordfolken