XCSoar/XCSoar #379: XCSoar map jitter if writing to Unix frame buffer

ubx, 2020-04-15:
I'm running XCSoar (version 7.0_preview15_Linux) on the following system:
Linux colibri-t30 3.1.10-2.8.6+g37b9f033652b #1 SMP PREEMPT Wed Mar 25 18:33:18 UTC 2020 armv7l GNU/Linux. XCSoar is compiled with USE_FB=y, i.e. it writes to the Linux frame buffer.

Everything works fine, except on the map I see a sort of image jitter as you can see on this
[video]( https://www.youtube.com/watch?v=AZSEHMfSoSM).

A simulation with same data on a "normal" Linux system (XCSoar compiled with OPENGL=y) this effect do not appear.

Any hints how to fix it?

lordfolken, 2020-04-17:
Seems to be a NVIDIA Tegra 3 chipset.

 Did you try the lxde image provided by the manufacturer i.e. with OPENGL accelerated drawing?

Found this:
https://developer.toradex.com/knowledge-base/framebuffer-(linux)

ubx, 2020-04-19:
Tried with this Toradex targets (exactly Colibri-T30_LXDE-Image_2.8b6.184-20190401):
angstrom-lxde-image
console-tdx-image
Always the same effect. To clarify: it happens only on the map windows [video2](https://www.youtube.com/watch?v=rgLxOkNaX2A).

For me, the most likely suspect is XCSoar.

lordfolken, 2020-04-19:
I have observed this now also in win64 with the latest preview15. When one sets the tracking to target the, map's elements and the gui overlay do not match up and start to move back and forth.

ubx, 2020-05-05:
Same problem with Ubuntu 20.04.

MaxKellermann, 2020-08-02:
This is caused by commit b0a3239019cc45e65581358b9f40258479ae0c9e
From https://github.com/XCSoar/XCSoar/pull/362 by @CptFrikadel

MaxKellermann, 2020-08-02:
(Sigh, this kind of regression happens every time, when I think "hey let's merge this even though it formally looks bad, at least people don't say I demotivate contributors with harsh criticism". A lot was wrong with this PR.)

closed 2020-08-02 by MaxKellermann