XCSoar/XCSoar #381: Build error: make target=UNIX USE_FB=y

ubx, 2020-04-21:
Investigating the issue #379 my intention is to reproduce the problem on my Ubuntu machine (Linux 5.3.0-46-generic #38-Ubuntu SMP, x86_64 GNU/Linux). 
Trying to compile XCSoar  (latest from master):

 make target=UNIX USE_FB=y

 I get the error:
```
 CXX     output/UNIX/dbg/src/HorizonWidget.o
In file included from ./src/Screen/Bitmap.hpp:32,
                 from src/Look/DialogLook.hpp:35,
                 from src/Look/Look.hpp:27,
                 from src/HorizonWidget.cpp:26:
./src/Screen/Memory/PixelTraits.hpp: In static member function ‘static constexpr BGRAPixelTraits::color_type* BGRAPixelTraits::NextByte(BGRAPixelTraits::pointer, int)’:
./src/Screen/Memory/PixelTraits.hpp:342:12: error: a reinterpret_cast is not a constant expression
  342 |     return pointer((uint8_t *)p + delta);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./src/Screen/Memory/PixelTraits.hpp: In static member function ‘static constexpr const color_type* BGRAPixelTraits::NextByte(BGRAPixelTraits::const_pointer, int)’:
./src/Screen/Memory/PixelTraits.hpp:347:12: error: a reinterpret_cast is not a constant expression
  347 |     return const_pointer((const uint8_t *)p + delta);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

lordfolken, 2020-05-04:
I'm sorry but the build environment supported is Debian 10. Currently both a docker container and a vagrant environment is provided at: https://github.com/XCSoar/XCSoar/tree/master/ide

closed 2020-05-04 by lordfolken

ubx, 2020-05-05:
Indeed, I was able to compile it on the docker environment.