XCSoar/XCSoar #383: Add display of GR required + terrain glide next to landables

bobmorane06, 2020-04-25:
Hello community, I cloned xcsoar locally and modified it to add one display mode for waypoints with glide ratio required + terrain glide – it works fine on android. 

The reason I did that is that I find it safer to make my in-flight navigation decisions based on GR (eg I decide not to go further than GR 20 from the closest landable on a glider with theoretical L/D of 40), but the arrival altitude adds confidence. Today it’s one or the other so I added an entry to the menu. 

I desperately looked for a forum to get help but https://forum.xcsoar.org is dead, subscribing to the mailing list didn’t work and no-one responded on the IRC. I am a noob and never used git / github , the last time I coded was 25 years ago, may I submit my change as a patch? How can I get directions to do it? 

[image]

rawtaz, 2020-04-25:
@bobmorane06 Welcome to GitHub then! :)

With all due respect, in response to "no-one responded on the IRC", you joined at (my time) 11:30, wrote two lines, then left at 11:31. Two minutes is certainly not enough time to get a response/help or to establish that no-one is responding in the channel. If you had sticked around and asked you might have gotten some directions.

I'd be willing to help you with the handling of Git/GitHub in general. I'm not an XCSoar developer but I do know the other stuff you need. Hit me up on the IRC, it's probably easier for me to guide you there than to hunt down guides on this topic.

bobmorane06, 2020-04-25:
Thanks, I'll ping you! I never used IRC BTW and I tried different IRC clients yesterday. I probably did it wrong. Sorry, didn't mean to be rude 🙂

rawtaz, 2020-04-25:
You did it just fine, your text came through and everything. Just do it the same way this time, but without closing the chat so quickly :P

PR #384 (2020-05-18): Renderer: New waypoint display: GR + arrival alt
This patch adds an additional mode for the display of “Arrival height” at a
waypoint, comprising both geometrical Glide Ratio and arrival altitude around
terrain.

Motivation: I find it safer to make my in-flight navigation decisions based
on GR (eg I decide not to go further than GR 20 from the closest landable on
a glider with theoretical L/D of 40), but the arrival altitude adds
confidence. In the current version it’s one or the other, this patch adds an
entry “Required GR & terrain glide” to the existing “None / Straight glide /
Terrain avoi [... cut]