XCSoar/XCSoar #386: Infobox not shown on Jotaphone 2 (Prev15, Android 5)

sasam, 2020-05-02:
v7.0-Android_prev15 have problem with infobox display, and flash red when changing display mods.
[image]

lordfolken, 2020-05-03:
MediaTek MT6580 Chipset
Mali-400 GPU

lordfolken, 2020-05-03:
Seems to be a driver issues with the device. The manufacturer refuses to fix. 
Here is another discussion about a imilar issue:
https://github.com/mrdoob/three.js/issues/9988

kedder, 2020-05-03:
This is a duplicate of #366. I have Yotaphone 2 myself and once bisected the issue to bff3273.

lordfolken, 2020-05-04:
I think XCSoar is currently not doing a good job of queering the GL extensions supported by the device. The api level change just forces the app to use a newer GLES version.

MaxKellermann, 2020-05-07:
To solve this, somebody needs an affected device and do a big debugging session with it. Without the device, nothing can be done.

MaxKellermann, 2020-06-13:
As I said: until somebody makes such a device available to a developer, nothing can be done. Closing this for now.

closed 2020-06-13 by MaxKellermann