XCSoar/XCSoar #388: XCSoar 6.8.13b-Android  "Apr 8 2020" on galaxy Tab4 Android 4.4.2. ==> display problem

Jonahan06340, 2020-05-07:
I installed XCSoar 6.8.13b-Android from "Apr 8 2020" on my galaxy Tab4 Android 4.4.2. I am having a display problem as shown in the screenshot below.
[image]
This setting that is responsible for this problem:
[image]
No porblem with other setting as below
[image]

MaxKellermann, 2020-05-07:
Did this work with an older XCSoar version?
This looks like a graphics driver bug on your device.

Jonahan06340, 2020-05-07:
It is the first time that I install XCsoar on my tablet. I never tried another version.

lordfolken, 2020-05-08:
@Jonahan06340 can you try some older builds? https://download.xcsoar.org/releases/ so we can see when this was introduced and if its your device or not.

Jonahan06340, 2020-05-08:
There same test:

6.8.11 Same Problem
6.8.5 Same Problem
6.8.1 Same Problem
6.7.7 Same Problem
6.7.3 Same Problem

Older Freeze my tab
6.7.1 Freeze
6.6.5 Freeze
6.6.1 Freeze
6.6 Freeze

Thx

lordfolken, 2020-05-11:
I cannot really reproduce it on any of my devices. Without access to a device, it will be hard to debug.

Jonahan06340, 2020-05-11:
If it helps, here is the display with the old versions when it freezes.
[image]

lordfolken, 2020-05-11:
Can you add the XCSoar-Log to the ticket please? You will find it the XCSoarData directory named XCSoar.log

Jonahan06340, 2020-05-12:
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/4613106/xcsoar.log)

[2020-05-11T18:23:26Z] Starting XCSoar XCSoar v6.8.11-Android
[2020-05-11T18:23:26Z] Display dpi=216,216
[2020-05-11T18:23:26Z] OpenGL: egl=1 oesdt=1 mda=1 npot=1 vbo=1 fbo=1 stencil=0x8d46
[2020-05-11T18:23:26Z] Initialise fonts
[2020-05-11T18:23:27Z] Startup dialog
[2020-05-11T18:23:28Z] Loading profiles
[2020-05-11T18:23:28Z] Loaded profile from /storage/emulated/0/XCSoarData/PEGASE.prf
[2020-05-11T18:23:28Z] Loading language file
[2020-05-11T18:23:28Z] Language: loading resource 'fr.mo'
[2020-05-11T18:23:28Z] Loaded translations from resource 'fr.mo'
[2020-05-11T18:23:28Z] Loading input events file
[2020-05-11T18:23:28Z] OpenTerrain
[2020-05-11T18:23:28Z] Loading Topography File...
[2020-05-11T18:23:28Z] ReadWaypoints
[2020-05-11T18:23:28Z] Failed to read waypoint file: /storage/emulated/0/XCSoarData/user.cup
[2020-05-11T18:23:28Z] Start at terrain center
[2020-05-11T18:23:28Z] RASP load
[2020-05-11T18:23:28Z] ReadAirspace
[2020-05-11T18:23:28Z] Failed to open airspace file: /storage/emulated/0/XCSoarData/AAPCA_Map.xcm/airspace.txt
[2020-05-11T18:23:29Z] Register serial devices
[2020-05-11T18:23:29Z] Opening device TCP port 4353
[2020-05-11T18:23:29Z] ProgramStarted
[2020-05-11T18:23:29Z] Free storage 67108864 KB

lordfolken, 2020-05-12:
Would you mind also uploading your profile and if possible an igc log?

Jonahan06340, 2020-05-12:
How to do this?

lordfolken, 2020-05-12:
The default.prf file and a log file xcsoar-*.igc

Jonahan06340, 2020-05-13:
[prf-icg.zip](https://github.com/XCSoar/XCSoar/files/4619579/prf-icg.zip)
Thx for your help

lordfolken, 2021-04-07:
Could you retest with 7.0?

Jonahan06340, 2021-04-08:
Where can i download it? link please?

lordfolken, 2021-07-02:
https://download.xcsoar.org/releases/ @Jonahan06340

MaxKellermann, 2021-07-09:
XCSoar 7 won't work because this is Android 4, and XCSoar now requires Android 5. Sorry, the device is unsupported now.

closed 2021-07-09 by MaxKellermann