XCSoar/XCSoar #389: New generation LX Navigation Device

K-Factor-1961, 2020-05-07:
I propose to develope a specific driver to communicate with all the new generation LX Navigation Devices.
In attach you can find the data port protocol for alla the new LX Navigation device as:
EOS57
EOS80
ERA57
ERA80
LX10K
COLIBRI X
I send you also a draft of ERA manual to understand how it works.
I'm also involved in feature developing for these devices.
[Era manual_rev_2.pdf](https://github.com/XCSoar/XCSoar/files/4594878/Era.manual_rev_2.pdf)

These devices in this moment exchange the protocol data to bluethooth port and to user port (NMEA RS232 port).
May be implemented bidirectional data interchange from device to external device connected with Xcsoar running.
[LX navigation communication protocols.pdf](https://github.com/XCSoar/XCSoar/files/4594852/LX.navigation.communication.protocols.pdf)
I can provide a EOS80 demo device to make tests
In the future may be possible also to use the LX Navigation Joy remote control mount on the glider stick to manage xcsoar app. 
In fact the LX Navigation Joy remote control can work in three different states. One of the states manages the Vario device, a second state could be used to manage the xcsoar app on the external device if you think usefull.

rawtaz, 2020-05-08:
@K-Factor-1961 Have you asked LX Navigation if they could contribute some code? It should be in their interest as well to get this integration working :)

K-Factor-1961, 2020-05-09:
> 
> 
> @K-Factor-1961 Have you asked LX Navigation if they could contribute some code? It should be in their interest as well to get this integration working :)

I asked them to develope this new dataport docoment to allow freware software develope team to realize new specific driver. This way all glider pilot can have a good solution to use Xcsoar in with new generation cheap smart vario EOS57, EOS80, ERA57, ERA80 LX10K. They send demo EOS80 version to test the dataport to the developer that need it. Ask me.

rawtaz, 2020-05-09:
Yes, but that's not what I asked. Just like anyone else in the world would spend their time writing an integration between LX stuff and XCSoar, why wouldn't they do it? It's just a question. I think it should be in their interest to have people using their hardware, and with that it should be in their interest to make their hardware work with one of the absolutely most used flight computer softwares out there. Did you ask them if they could write some or all of the code?

lordfolken, 2020-05-10:
Marco send me a mail to folken@kabelsalat.ch, we can talk about the implementation.

K-Factor-1961, 2020-05-12:
> Yes, but that's not what I asked. Just like anyone else in the world would spend their time writing an integration between LX stuff and XCSoar, why wouldn't they do it? It's just a question. I think [...]

I understand your question. Right now LX Navigation is very committed to the development of instruments that are all new generation. They have to make up for a major delay in development because of the few human resources available in development in relation to the amount of work.
So they have no resources for this right now.
But they are available to provide demo units through my coordination.
I am only the dealer for Italy, although an aerospace engineer, I am above all a glider pilot. I managed to make at least LX Navigation produce and implement in the FW of its DATA PORT DOCUMENT devices. Many developers were very interested in this. Once I got the standard I thought to distribute it to everyone to allow the glider pilot community to benefit from the performance of the new devices on open source navigation software.

rawtaz, 2020-05-12:
@K-Factor-1961 Thanks for clarifying!

vyzaq, 2020-05-24:
I will be happy to help with LX Navigation driver, I have one Era 80 installed in my glider and I really need the driver to exchange MC, Bugs, etc information. I'm a newbie in XCSoar but have some experience in C++, will appreciate any help on that.

lordfolken, 2020-07-28:
[LX_CP_R2.pdf](https://github.com/XCSoar/XCSoar/files/4988271/LX_CP_R2.pdf)
Revsion 2 of the document.

lordfolken, 2020-07-28:
https://github.com/lordfolken/XCSoar/tree/lxera experimental implementation.

vyzaq, 2020-07-28:
also close to finishing implementation, with updated datasheet thing will go faster, hope they fixed issues with protocol.
https://github.com/vyzaq/XCSoar/tree/lx_navigation_driver

lordfolken, 2020-08-05:
[LX_CP_R3.pdf](https://github.com/XCSoar/XCSoar/files/5031405/LX_CP_R3.pdf)

lordfolken, 2020-08-13:
[LX_CP_R4.pdf](https://github.com/XCSoar/XCSoar/files/5069941/LX_CP_R4.pdf)

lordfolken, 2020-08-13:
@vyzaq why don't you make a pr request and put in draft? then we can work together.

vyzaq, 2020-12-23:
sorry for a long silence.
I just got back my device after the season and can finalize my efforts with flights logs reading, I need to validate several bugs that were in previous firmware.
@lordfolken sure, It will be great, I'm not GitHub pro contributor, so need some guidance on how to work together on the same pull request :)

Dorianosaur, 2023-05-26:
Was this driver issue ever resolved? I got my era 80 about 2 weeks ago and there's still no driver in XC soar

lordfolken, 2023-05-29:
No, the firmware bugs are still there. Xcsoar cannot support a device that doesn't supply a correct altitude/ qnh.

MaxKellermann, 2023-05-29:
I repeat: the ERA80 is **NOT** supported by XCSoar.

Dorianosaur, 2023-05-30:
> No, the firmware bugs are still there. Xcsoar cannot support a device that doesn't supply a correct altitude/ qnh.
> 
> If you plan to use xcsoar check first for device support, before you buy.

Ah... In my defence, it is advertised as working with XC soar by Lx Navigation. Strangely the LXNAV driver appears to work somewhat - I can send the declaration from XC soar to the Era, but it throws an error on the XC soar side

MaxKellermann, 2023-05-30:
> it is advertised as working with XC soar by Lx Navigation

They do? If that is true, it could probably make you eligible for a refund. Better double-check what kind of comaptibility was really promised.
I can't find anything on the net. All Google gives me is https://github.com/XCSoar/XCSoar/issues/415 which clearly states that the ERA80 is **not** supported by XCSoar.

In order to avoid confusing LX Navigation with LXNAV (who the .... came up with these competing company names??), we renamed the ambiguously named "LX" driver to "LXNAV" 8 years ago, so people know the driver supports LXNAV, but not LX Navigation products.

rawtaz, 2023-05-30:
Everyone should stop buying LX Navigation products anyway. Worst company ever. Use LXNAV stuff instead.

MaxKellermann, 2023-05-31:
> Worst company ever

No, I wouldn't say that. Before XCSoar became popular and mainstream, they made some very incompetent choices and annoyed the hell out of me. When they realized that XCSoar support was indeed important for their customers, they tried over and over to convince me to bring support for their products, but the damage was done and I decided I don't want to spend my free time to help them make money (adding a hundred thousand potential customers). They still had the choice to just write the required XCSoar code and submit them to us, but they decided they don't want to spend their time on XCSoar, so .... that's it.

I guess their products are decent, but without XCSoar support, I wouldn't buy any. Connecting all the sensors in your cockpit is important - it's the one initial reason why I joined XCSoar development 14 years ago. Not being open about your communication protocols is a no-go.

Dorianosaur, 2023-05-31:
Their site is currently down but it did state on the Era page that it's compatible with both XC soar and CU. Here's a link to the Era manual about connecting XC soar https://www.manualslib.com/manual/1944880/Lx-Navigation-Era-Series.html?page=76#manual

jirimaier, 2023-07-15:
Hello,

from my experience with LX Eos, the LXNAV driver is working quite well for the LX Navigation device. The only (and quite significant) issue is the discrepancy in altitude data (LX Navigation provides true altitude, while LXNAV uses standard altitude).

By looking at the source code, I think it may be quite straightforward to fix this:
The LXNAV parser calls `info.ProvidePressureAltitude(value)`.
There also exists a method `info.ProvideBaroAltitudeTrue(value)`, which is used in the Condor driver.

Would it be possible to simply make a copy of the LXNAV driver and change this one line of code?
Of course, we wouldn't be able to confidently say that it is compatible with all LX Navigation devices, but it would solve the most significant incompatibility and can be good enough as an "experimental" driver.

MaxKellermann, 2023-07-15:
> Would it be possible to simply make a copy of the LXNAV driver and change this one line of code?

That's possible, but it would work properly only with certain LX Navigation products with certain firmware versions. XCSoar doesn't support such inconsistent behavior. Don't adapt XCSoar to broken products, don't buy broken products.

And anyway, duplicate code will not be merged into XCSoar.

What LX Navigation could do, if they feel XCSoar support is important, is to write a driver which works well with all of their products with all firmware versions and submit it to us, with a promise to keep maintaining it.

Until that happens, or until somebody else volunteers to develop and maintain such a driver, let's not speculate about what some unknown person could do. This person doesn't exist currently. @jirimaier, do you volunteer?

jirimaier, 2023-07-16:
Volunteering to maintain a driver for all LX Navigation products would be too much for me. But I can make a driver specifically for the LX Eos and maintain it. Hopefully, it will work with other similar devices (like Era #415) too, but that won't be guaranteed.

MaxKellermann, 2023-07-16:
A driver just for the Eos would be fine, but remember that duplicate code is not acceptable and will not be merged. If you want it merged here, do it in a way that doesn't duplicate any code.

closed 2024-12-07 by lordfolken