XCSoar/XCSoar #390: Version 6.8.13b does not open jpg/pdf pictures in the image viewer on Samsung Galaxy Note10+ / Android10

Buzz33, 2020-05-07:
In the waypoint description file it is possible to implement links to pictures. If you click now the file path of the associatetd picture (e.g. a pdf file of an outlanding site, see screenshot - second line) the line will be highlited for a short moment, but then nothing happens. The picture viewer does not start, the picture is not shown on the display.

[image]

This happens with Version 6.8.13b and 6.8.12 on Android 10 Smartphones.
The older Version 6.8.11 works fine in this point on my Note 10+ with Android 10!

On my Samsung Galaxy Note 2 with Android 4.4.2 and on my Samsung Galaxy Grand Prime with Android 5.1.1 all above mentioned XCSoar Versions work fine in this point.

Seems to be a problem that starts with Version 6.8.12 in relationship with higher Android Versions.

Buzz33, 2020-06-24:
This Bug persists also in Version 6.8.15.

So I cannot use my outlanding catalogue in XCSoar with Version. 6.8.12 to 6.8.15. 
Because this is a key Feature to me I have to stay with Version 6.8.11, despite the Installing dialoge tells me that Version 6.8.11 ist too old for my Android Version (but 6.8.11 seems to work so far with Android10).

lordfolken, 2020-07-02:
https://github.com/XCSoar/XCSoar/blob/4cd3f8fa9e4f8966a59ed5f0ffc86de11c3858dd/android/src/NativeView.java#L385
The fileOpen() Function is responsible for finding and starting the correct application for the correct file type. The function wasn't changed in years, so either the function is no longer called, or something in the function broke when the android ndk version was incremented or the permission handling was changed. 

The function should log the error NativeView ... error if its called and failed. 

One thing that bothers me is that the mimetype is discovered from the file extension. This isn't exactly good practice.

Buzz33, 2020-07-12:
Just testet it again: 

There  is no failure message  poping up on the display after calling and failing. 
Also in the XCSoar log file there is no NativeView... error recorded.
 
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/4909296/xcsoar.log)

Buzz33, 2020-08-14:
This bug persists in Version 6.8.16 too.

leojullerot, 2020-09-27:
Bug persists in V6.8.17

lordfolken, 2020-09-27:
Got the error with logcat:
```
09-28 00:47:11.314  9546  9567 E XCSoar  : NativeView.openFile('/storage/emulated/0/XCSoarData/lspd.pdf') error
09-28 00:47:11.314  9546  9567 E XCSoar  : android.os.FileUriExposedException: file:///storage/emulated/0/XCSoarData/lspd.pdf exposed beyond app through Intent.getData()
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.os.StrictMode.onFileUriExposed(StrictMode.java:1978)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.net.Uri.checkFileUriExposed(Uri.java:2371)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.content.Intent.prepareToLeaveProcess(Intent.java:10260)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.content.Intent.prepareToLeaveProcess(Intent.java:10213)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1669)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.app.Activity.startActivityForResult(Activity.java:4586)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.app.Activity.startActivityForResult(Activity.java:4544)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.app.Activity.startActivity(Activity.java:4905)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at android.app.Activity.startActivity(Activity.java:4873)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at org.xcsoar.NativeView.openFile(NativeView.java:410)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at org.xcsoar.NativeView.runNative(Native Method)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at org.xcsoar.NativeView.run(NativeView.java:295)
09-28 00:47:11.314  9546  9567 E XCSoar  : 	at java.lang.Thread.run(Thread.java:764)
```
This breakage is introduced by the changes in android 7.0;
https://developer.android.com/about/versions/nougat/android-7.0-changes

A solution:
https://stackoverflow.com/questions/38200282/android-os-fileuriexposedexception-file-storage-emulated-0-test-txt-exposed

Buzz33, 2021-06-05:
Milestone was 7.4 - Outlanding Pictures still not usable in 7.4 and 7.7.  :-( 
The good news: The 2 years old 6.8.11 still works with Anfroid 11 ;-)

lordfolken, 2021-06-05:
Well you didn't provide a PR. :-)
https://github.com/XCSoar/XCSoar/blob/fce4c0a04a622037705a0c14de484c5b40eaa045/android/src/NativeView.java#L427
is the function that is bugging out.

Buzz33, 2021-06-06:
>>Well you didn't provide a PR. :-)<<

Thank you for your investigation so far. :-)
But sorry, I am just a dumb user that can report issues and can do some testing. Would have tried it by myself  for quite some time if I had the skills. 

Maybe anybody else can help?

lordfolken, 2021-06-06:
Iplayed around a bit and got at least the intent call working. So it tries to open the file with an app I
E. Pdf. The app doesn't get the correct path however.

lordfolken, 2021-06-06:
Maybe I could interest you in #242 ?

Buzz33, 2021-06-06:
>> Iplayed around a bit and got at least the intent call working. <<

So there will be hope !? :-))

>>Maybe I could interest you in #242 ?<<

Well, I can try to write some text and make screenshots for the manual. 

Some questions for that: 

Shall I describe the functionality and layout from Version 6.8.11 then? Or are there any changes in the WP-Description File functionality or in the layout since the 6.8.11 in the current 7.8 (despite the picture bug #390) ? 

What format shall I use? Word, PP, pdf ?
Or just txt or rtf? Then with the added jpg, png, tif or bmp pictures in a seperated folder?

Buzz33, 2021-06-19:
Maybe these where too much stupid questions... (?) ;-)

OK, yesterday, a friend of mine gave me a most basic crash course in git - and we fiddlet about where and how things seem to be organised relating to the manual.
Despite my knowledge with git and .tex currently are in the most basic way, I will try to get the manual updated in some points - about the waypoint description file, use of pictures and mediafiles etc. But it will take some time...

closed 2021-09-07 by MaxKellermann