XCSoar/XCSoar #393: Flight log date from LX nano is wrong

condoratberlin, 2020-05-10:
I downloaded a flight from my nano logger done at the 9.5.2020. XCSOAR shows the flight by using the flight download menue with a wrong date (year). It shows 9.5.2010.
The file is correct and the file name was also correct. The flight was approved by OLC without any problems.
The log from my flarm was shown with the correct date! So it seems to be an issue with the LX driver?

lordfolken, 2020-05-10:
Are you using at least Firmware 3.00 on your Nano?
https://gliding.lxnav.com/lxdownloads/firmware/

Changelog there:
Version 3.00
    Year 2020 fix for nanoconfig app.
Version 2.56
    GPS rollover bug fix
Version 2.50
    Fixed problem for GPS rollover (April 2019)
Version 2.45
    Fixed compatibility issues with nanoconfig for android

What does the nanoconfig app say when you connect it to the nano directly?
The igc logfile from XCSoar would also be interesting...
What version of XCSoar on what platform?
Debug log from XCSoar?

condoratberlin, 2020-05-10:
I will check tomorow, but I haven't update the nano firmware for a while. So I doubt to have the 3.00.

condoratberlin, 2020-05-11:
Ok, the Firmware version of my Nano was still at 2.50. After updating to 3.00 the flight log dates were shown correctly. So it is was an issue of my nano logger.
I close the issue.

closed 2020-05-11 by condoratberlin