XCSoar/XCSoar #397: 30s average Needle is not inverted by setting "Invert Info boxes"

condoratberlin, 2020-05-11:
Inverting the infoboxes gives them a black background. The current climb needle is inverted to white then, but the 30s average needle (trinagle) is not and can barely be seen as shown here:
[image]

lordfolken, 2020-05-11:
https://github.com/XCSoar/XCSoar/blob/719cedbf614b037c75081fa86f7be3735fb0daf4/src/Gauge/GaugeVario.cpp#L295

Seems to be the code for selecting the color of the needle. 

I don't understand the notation form there, but the one for the varioneedle defined above: 
https://github.com/XCSoar/XCSoar/blob/719cedbf614b037c75081fa86f7be3735fb0daf4/src/Gauge/GaugeVario.cpp#L278
Seems more logical too me.

condoratberlin, 2020-05-12:
For the vario needle and the average needle the same method is used. the average bool value divides between the two needles.
For the vario needle it works correct.

condoratberlin, 2020-05-12:
I think the difference is in the canvas methods:
https://github.com/XCSoar/XCSoar/blob/719cedbf614b037c75081fa86f7be3735fb0daf4/src/Gauge/GaugeVario.cpp#L301-L304
For the vario needle the following method is used
https://github.com/XCSoar/XCSoar/blob/351928bdfb56bfd7237a1abe8711c18171827f3d/src/Screen/OpenGL/Canvas.cpp#L209-L239
where an brush and pen are used depending on number of Points. and if a brush is not hollow.
The method used for the average needle is:
https://github.com/XCSoar/XCSoar/blob/351928bdfb56bfd7237a1abe8711c18171827f3d/src/Screen/OpenGL/Canvas.cpp#L156-L167
where only a pen is used. But the only the brush is inverted.

condoratberlin, 2020-05-12:
When inserting a 
`canvas.SelectWhitePen();`
in addition to canvas.SelectWhiteBrush
https://github.com/XCSoar/XCSoar/blob/351928bdfb56bfd7237a1abe8711c18171827f3d/src/Gauge/GaugeVario.cpp#L295-L297 
the average needle will be in white.
[image]
I will make some more tests an create a pullrequest.

condoratberlin, 2020-05-13:
Ok, I will create a patch for the v6.8. branch. Is it possible to combine both pill requests or does it have to be another pull request?

lordfolken, 2020-05-14:
Thank you two for all the effort!

PR #400 (2020-05-14): Fixes 30s average needle color when boxes are inverted for v6.8.xx
fixes #397 The average needle is painted with a pen instead a brush, therefore also the pen color is selected for V6.8.xx.
commit 6bcfd3f0 (2020-05-14): fixes #397 The average needle is painted with a pen instead a brush, therefore also the pen color is selected.
closed 2020-05-14 by lordfolken