XCSoar/XCSoar #401: XCSoar 6.8.13 Kobo Glo HD + BlueFly Vario = crash / Close

TokkieC, 2020-05-14:
When using the latest XCSoar 6.8.13 (release) for Kobo with a BlueFly Vario in "FLY" mode the program closes / crashes after some time.
It does not look like the GPS data is coming through / being updated...

Loading 6.8.11 all is good again...

lordfolken, 2020-05-14:
Can you provide us with the xcsoar.log ?

zvratko, 2020-05-14:
the same problem with kobo mini, where can i fing log? There is no log in Log File.

zvratko, 2020-05-14:
there is log from 2011.

zvratko, 2020-05-14:
it crashes immediatelly GPS lock possition

TokkieC, 2020-05-14:
xcsoar.log:

[2020-05-14T18:20:40Z] Free storage 6900424 KB
[2020-05-14T18:20:40Z] Starting XCSoar XCSoar v6.8.13-Kobo
[2020-05-14T18:20:40Z] Display dpi=300,300
[2020-05-14T18:20:40Z] Initialise fonts
[2020-05-14T18:20:40Z] Startup dialog
[2020-05-14T18:20:40Z] Loading profiles
[2020-05-14T18:20:40Z] Loaded profile from /mnt/onboard/XCSoarData/default.prf
[2020-05-14T18:20:41Z] Display rotated
[2020-05-14T18:20:41Z] Initialise fonts
[2020-05-14T18:20:41Z] Loading language file
[2020-05-14T18:20:41Z] Loading input events file
[2020-05-14T18:20:41Z] OpenTerrain
[2020-05-14T18:20:41Z] Loading Topography File...
[2020-05-14T18:20:41Z] ReadWaypoints
[2020-05-14T18:20:41Z] Failed to read waypoint file: /mnt/onboard/XCSoarData/user.cup
[2020-05-14T18:20:41Z] RASP load
[2020-05-14T18:20:41Z] ReadAirspace
[2020-05-14T18:20:42Z] Register serial devices
[2020-05-14T18:20:42Z] Opening device /dev/ttymxc0
[2020-05-14T18:20:42Z] ProgramStarted
[2020-05-14T18:20:42Z] Free storage 6900420 KB
[2020-05-14T18:21:33Z] Saving profiles
[2020-05-14T18:21:33Z] Saving profile to /mnt/onboard/XCSoarData/default.prf
[2020-05-14T18:21:42Z] ReadWaypoints
[2020-05-14T18:21:42Z] Failed to read waypoint file: /mnt/onboard/XCSoarData/user.cup
[2020-05-14T18:21:42Z] Start at terrain center
[2020-05-14T18:21:42Z] Loading Topography File...
[2020-05-14T18:21:43Z] ReadAirspace
[2020-05-14T18:21:43Z] Failed to open airspace file: /mnt/onboard/XCSoarData/Bloemfontein.xcm/airspace.txt 

default.prf:

AdditionalWPFile=""
AirspaceFile="%LOCAL_PATH%\Airspace/RSA_Airspace_17AUG2017.txt"
AltitudeUnit="10"
AppAveNeedle="0"
AppGaugeVarioGross="0"
AppIndLandable="1"
AutoWind="3"
AutoZoom="0"
AverEffTime="2"
ClimbMapScale="9574"
CruiseMapScale="319"
DeviceA="BlueFly"
DeviceB="Borgelt B50"
DisplayOrientation="0"
DrawTerrain="0"
DrawTopology="1"
EnableFLARMDisplay="0"
EnableFLARMGauge="0"
EnableFlightLogger="1"
EnableNMEALogger="0"
EnableVarioBar="0"
ExternalWind="1"
FinishType="12"
FriendsBlue=""
FriendsGreen=""
FriendsMagenta=""
FriendsYellow=""
HomeLocation="26.169167 -29.035667"
HomeWaypoint="-1"
InfoBoxPanel1Box0="10"
InfoBoxPanel1Box1="69"
InfoBoxPanel1Box10="0"
InfoBoxPanel1Box11="0"
InfoBoxPanel1Box12="0"
InfoBoxPanel1Box13="0"
InfoBoxPanel1Box14="0"
InfoBoxPanel1Box15="0"
InfoBoxPanel1Box16="0"
InfoBoxPanel1Box17="0"
InfoBoxPanel1Box18="0"
InfoBoxPanel1Box19="0"
InfoBoxPanel1Box2="94"
InfoBoxPanel1Box20="0"
InfoBoxPanel1Box21="0"
InfoBoxPanel1Box22="0"
InfoBoxPanel1Box23="0"
InfoBoxPanel1Box3="1"
InfoBoxPanel1Box4="14"
InfoBoxPanel1Box5="11"
InfoBoxPanel1Box6="6"
InfoBoxPanel1Box7="17"
InfoBoxPanel1Box8="5"
InfoBoxPanel1Box9="0"
InputFile=""
LanguageFile="none"
LoggerID="GOG"
LoggerTimeStepCruise="3"
MapFile="%LOCAL_PATH%\Bloemfontein.xcm"
MaxAutoZoomDistance="30000.000000"
OrientationCruise="0"
Page1Bottom="0"
Page1InfoBoxMode="0"
Page1InfoBoxPanel="0"
Page1Layout="0"
Page1Main="0"
PilotName="TC"
PlanePath="%LOCAL_PATH%\ZS-GOG.xcp"
Port2BaudRate="4800"
Port2BluetoothMAC=""
Port2BulkBaudRate="0"
Port2Enabled="0"
Port2I2C_Addr="0"
Port2I2C_Bus="2"
Port2IOIOUartID="0"
Port2IPAddress=""
Port2K6Bt="0"
Port2Path="/dev/ttyUSB0"
Port2PressureUse="2"
Port2SecondDevice="Generic"
Port2SensorFactor="0.000000"
Port2SensorOffset="0.000000"
Port2SyncFromDevice="1"
Port2SyncToDevice="1"
Port2TCPPort="4353"
Port2Type="serial"
Port2UseSecondDevice="0"
PortBaudRate="57600"
PortBluetoothMAC=""
PortBulkBaudRate="0"
PortEnabled="1"
PortI2C_Addr="0"
PortI2C_Bus="2"
PortIOIOUartID="0"
PortIPAddress=""
PortK6Bt="0"
PortPath="/dev/ttymxc0"
PortPressureUse="2"
PortSecondDevice=""
PortSensorFactor="0.000000"
PortSensorOffset="0.000000"
PortSyncFromDevice="1"
PortSyncToDevice="1"
PortTCPPort="4353"
PortType="serial"
PortUseSecondDevice="0"
SafetyAltitudeArrival="152.400000"
SafetyAltitudeTerrain="210.000000"
SafetyMacCready="10"
SetSystemTimeFromGPS="1"
SlopeShadingType="3"
StartType="1"
TeamcodeRefWaypoint="-1"
TerrainBrightness="36"
TerrainContours="1"
TerrainContrast="151"
TerrainRamp="1"
UTCOffsetSigned="7200"
UserLevel="1"
WPFile="%LOCAL_PATH%\Turnpoints/NewTempe-Turnpoints.cup"
WaveAssistant="0"
WayPointLabelSelection="0"

TokkieC, 2020-05-14:
I am only able to open XCsoar into "FLY" once after a cold start now. As soon as it crash it won'y open again until after a shutdown.
It also crash before even starting the Bluefly Vario currently.

The airspace, Map and turnpoint files work fine with 6.8.11

MaxKellermann, 2020-05-14:
This is most likely caused by 7aee59bbee376eaa60d94639e22c762ad60c535a
Musl uses a default stack size of 80 kB, but Jasper creates a 256 kB buffer on the stack.

closed 2020-05-14 by MaxKellermann

cefn, 2020-06-13:
Hi, all. I have a Kobo+BlueFly which experienced the same problem. It was previously fine with 6.8.6

I just installed 6.8.15 from https://download.xcsoar.org/releases/6.8.15/KOBO/ by dragging the KoboRoot.tgz to the hidden `.kobo` folder and restarting. 

I can confirm that it now loads in both Sim and Fly mode without crashing.