XCSoar/XCSoar #402: Not fullscreen with Android Q Immersive mode on 18:9 screen

infeeeee, 2020-05-15:
When I open XCSoar a huge black bar opens at the bottom, and it seems the screen shows up as 16:9, but my phone has a 18:9 screen:

[image]

The black bar stays when I start the app.

Both stable and testing affected.

I use immersive fullscreen mode. It means I don't have virtual or physical control buttons, I use gesture navigation. 

Some roms have a feature "legacy app scaling" or similar, which solves this issue. But I think this problem should be addressed on the app side, not rom side, every other app on my phone scales correctly.

As more and more phones have a non 16:9 screen, I think it will affect more people in the future.

I tried to solve this manually with `adb shell settings put global policy_control immersive.full=...` but it didn't help (maybe it's a rom problem, not with the app, but as I said, normally I shouldn't need this)

Device: Xiaomi Mi Mix 3
ROM: Android Q, ArrowOS 10.0 Official

lordfolken, 2020-05-16:
Hmm, we fixed this already, so I'm a bit surprised.
https://github.com/XCSoar/XCSoar/commit/fb8370c2ed5df0e69a1055e3d53d922847ae2ea4#diff-9832a7bd42037dd9b6595bb2c5eea58b

XCSoar can take pretty much any resolution so codewise it shouldn't be a problem.

infeeeee, 2020-05-17:
I don't know if it helps, but here is my log saved with matlog libre. I filtered for xcsoar, cleared before I started xcsoar-testing:
```
05-17 16:18:36.336 I/ActivityTaskManager(1508): START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=org.xcsoar.testing/.XCSoar bnds=[239,407][440,726]} from uid 10197
05-17 16:18:36.399 I/ActivityManager(1508): Start proc 12547:org.xcsoar.testing/u0a217 for activity {org.xcsoar.testing/org.xcsoar.testing.XCSoar}
05-17 16:18:36.406 E/.xcsoar.testin(12547): Not starting debugger since process cannot load the jdwp agent.
05-17 16:18:36.416 W/.xcsoar.testin(12547): Core platform API violation: Ljava/lang/reflect/Field;->accessFlags:I from Landroid/os/Build; using reflection
05-17 16:18:36.531 D/XCSoar  (12547): ABI=arm64-v8a
05-17 16:18:36.531 D/XCSoar  (12547): PRODUCT=perseus
05-17 16:18:36.531 D/XCSoar  (12547): MANUFACTURER=Xiaomi
05-17 16:18:36.531 D/XCSoar  (12547): MODEL=Mi MIX 3
05-17 16:18:36.531 D/XCSoar  (12547): DEVICE=perseus
05-17 16:18:36.531 D/XCSoar  (12547): BOARD=sdm845
05-17 16:18:36.531 D/XCSoar  (12547): FINGERPRINT=Xiaomi/perseus/perseus:10/QKQ1.190828.002/V11.0.3.0.QEEMIXM:user/release-keys
05-17 16:18:36.545 D/XCSoar  (12547): getExternalStorageState() = mounted
05-17 16:18:36.571 D/XCSoar  (12547): EGL vendor: Android
05-17 16:18:36.571 D/XCSoar  (12547): EGL version: 1.5 Android META-EGL
05-17 16:18:36.571 D/XCSoar  (12547): EGL extensions: EGL_KHR_get_all_proc_addresses EGL_ANDROID_presentation_time EGL_KHR_swap_buffers_with_damage EGL_ANDROID_get_native_client_buffer EGL_ANDROID_front_buffer_auto_refresh EGL_ANDROID_get_frame_timestamps EGL_EXT_surface_SMPTE2086_metadata EGL_EXT_surface_CTA861_3_metadata EGL_KHR_image EGL_KHR_image_base EGL_EXT_image_gl_colorspace EGL_KHR_lock_surface EGL_KHR_gl_colorspace EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_gl_renderbuffer_image EGL_KHR_reusable_sync EGL_KHR_fence_sync EGL_KHR_create_context EGL_KHR_surfaceless_context EGL_EXT_create_context_robustness EGL_ANDROID_image_native_buffer EGL_KHR_wait_sync EGL_ANDROID_recordable EGL_KHR_partial_update EGL_EXT_pixel_format_float EGL_KHR_create_context_no_error EGL_KHR_mutable_render_buffer EGL_EXT_yuv_surface EGL_EXT_protected_content EGL_IMG_context_priority EGL_KHR_no_config_context 
05-17 16:18:36.572 D/XCSoar  (12547): EGLConfig = {rgba=5/6/5/0; depth=0; stencil=8}
05-17 16:18:36.572 D/XCSoar  (12547): OpenGL vendor: Qualcomm
05-17 16:18:36.572 D/XCSoar  (12547): OpenGL version: OpenGL ES 3.2 V@415.0 (GIT@775f2c1, I0e419467bc, 1577280788) (Date:12/25/19)
05-17 16:18:36.573 D/XCSoar  (12547): OpenGL renderer: Adreno (TM) 630
05-17 16:18:36.573 D/XCSoar  (12547): OpenGL extensions: GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_half_float GL_OES_framebuffer_object GL_OES_rgb8_rgba8 GL_OES_compressed_ETC1_RGB8_texture GL_AMD_compressed_ATC_texture GL_KHR_texture_compression_astc_ldr GL_KHR_texture_compression_astc_hdr GL_OES_texture_compression_astc GL_OES_texture_npot GL_EXT_texture_filter_anisotropic GL_EXT_texture_format_BGRA8888 GL_EXT_read_format_bgra GL_OES_texture_3D GL_EXT_color_buffer_float GL_EXT_color_buffer_half_float GL_QCOM_alpha_test GL_OES_depth24 GL_OES_packed_depth_stencil GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_EXT_sRGB GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_EXT_texture_type_2_10_10_10_REV GL_EXT_texture_sRGB_decode GL_EXT_texture_format_sRGB_override GL_OES_element_index_uint GL_EXT_copy_image GL_EXT_geometry_shader GL_EXT_tessellation_shader GL_OES_texture_stencil8 GL_EXT_shader_io_blocks GL_OES_shader_image_atomic GL_OES_sample_variables GL_EXT_texture_border_clamp GL_EXT_EGL_image_external_wrap_modes GL_EXT_multisampled_render_to_texture GL_EXT_multisampled_render_to_texture2 GL_OES_shader_multisample_interpolation GL_EXT_texture_cube_map_array GL_EXT_draw_buffers_indexed GL_EXT_gpu_shader5 GL_EXT_robustness GL_EXT_texture_buffer GL_EXT_shader_framebuffer_fetch GL_ARM_shader_framebuffer_fetch_depth_stencil GL_OES_texture_storage_multisample_2d_array GL_OES_sample_shading GL_OES_get_program_binary GL_EXT_debug_label GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent GL_QCOM_tiled_rendering GL_ANDROID_extension_pack_es31a GL_EXT_primitive_bounding_box GL_OES_standard_derivatives GL_OES_vertex_array_object GL_EXT_disjoint_timer_query GL_KHR_debug GL_EXT_YUV_target GL_EXT_sRGB_write_control GL_EXT_texture_norm16 GL_EXT_discard_framebuffer GL_OES_surfaceless_context GL_OVR_multiview GL_OVR_multiview2 GL_EXT_texture_sRGB_R8 GL_KHR_no_error GL_EXT_debug_marker GL_OES_EGL_image_external_essl3 GL_OVR_multiview_multisampled_render_to_texture GL_EXT_buffer_storage GL_EXT_external_buffer GL_EXT_blit_framebuffer_params GL_EXT_clip_cull_distance GL_EXT_protected_textures GL_EXT_shader_non_constant_global_initializers GL_QCOM_texture_foveated GL_QCOM_texture_foveated_subsampled_layout GL_QCOM_shader_framebuffer_fetch_noncoherent GL_QCOM_shader_framebuffer_fetch_rate GL_EXT_memory_object GL_EXT_memory_object_fd GL_EXT_EGL_image_array GL_NV_shader_noperspective_interpolation GL_KHR_robust_buffer_access_behavior GL_EXT_EGL_image_storage GL_EXT_blend_func_extended GL_EXT_clip_control GL_OES_texture_view GL_EXT_fragment_invocation_density GL_QCOM_YUV_texture_gather 
[... cut ...]
05-17 16:18:40.815 E/InputDispatcher(1508): channel '3ffff51 org.xcsoar.testing/org.xcsoar.testing.XCSoar (server)' ~ Channel is unrecoverably broken and will be disposed!
05-17 16:18:40.817 I/ActivityManager(1508): Process org.xcsoar.testing (pid 12547) has died: cch CRE 
05-17 16:18:40.817 I/WindowManager(1508): WIN DEATH: Window{3ffff51 u0 org.xcsoar.testing/org.xcsoar.testing.XCSoar}
05-17 16:18:40.817 W/InputDispatcher(1508): Attempted to unregister already unregistered input channel '3ffff51 org.xcsoar.testing/org.xcsoar.testing.XCSoar (server)'
```

johnthomas22, 2020-05-29:
> Hmm, this was fixed so I'm a bit surprised...
> [fb8370c#diff-9832a7bd42037dd9b6595bb2c5eea58b](https://github.com/XCSoar/XCSoar/commit/fb8370c2ed5df0e69a1055e3d53d922847ae2ea4#diff-9832a7bd42037dd9b6595bb2c5eea58b)
> 
> XCSoar can take pretty much any resolution so code-wise it shouldn't be a problem.

If you mean the fix aimed at the Pixel 4 and other devices with taller screens, the fix did not work. The manifest was updated by another member of my club, but it did not fix the problem on his phone, nor mine. Here's hoping... 😁

lordfolken, 2021-06-09:
@johnthomas22 is this still a  problem with 7.8?

johnthomas22, 2021-06-09:
Philipp,

In the sense it's still noticeable, yes, though I have no evidence it's
obscuring anything that matters.

7.8, exciting, will have to see if the other things are fixed - GPS
tracking in background and hanging menu in waypoint details.

Regards

John Thomas

lordfolken, 2021-06-09:
#618 is fixed so you should have background gps

johnthomas22, 2021-06-09:
Great, can't verify until my next flight but  that's good to know.

I can confirm https://github.com/XCSoar/XCSoar/issues/638 is fixed though.
Nice one, thanks to Max for picking it up so quickly.

Regards,

John Thomas