XCSoar/XCSoar #404: Live Tracking not possible in V6.8.14

chz007, 2020-05-16:
In the current version 6.8.14, the live track is not found, even though Live Tracking is switched on for Skylines.
A new Live Tracking Key has already been generated and tested.
With the current XCSoar testversion 7.0_preview15-Android the Live Track is visible.
System: OnePlus 6t with Android OxygenOS 10.3.3 (Android 10)

dernatsch, 2020-05-18:
7.0_preview15 uses a DNS lookup 
https://github.com/XCSoar/XCSoar/blob/0628a6812cf80ce0795f37e997af160a576ac88c/src/Tracking/SkyLines/Glue.cpp#L218-L220
whereas v6.8.14 uses a harcoded IP
https://github.com/XCSoar/XCSoar/blob/8e0f36a3027efa452437c3d7f4c1dd4728cd3814/src/Tracking/SkyLines/Glue.cpp#L223

I think this could be the cause, as nslookup gives `212.51.149.149` for tracking.skylines.aero which differs from the hardcoded one

lordfolken, 2020-05-18:
Didn't see that this was fixed only in master. I had to move skylines.aero on an emergency basis.

lordfolken, 2020-05-18:
https://github.com/XCSoar/XCSoar/blob/0628a6812cf80ce0795f37e997af160a576ac88c/src/Tracking/SkyLines/Glue.cpp#L218-L220 backport needed.

MaxKellermann, 2020-05-18:
No, v6.8.x doesn't have `boost::asio`, can't backport that. We need to hard-code the new IP address for that branch.

dernatsch, 2020-05-18:
opened PR #409

lordfolken, 2020-05-19:
There is some irony in this, in that this issue has number #404...

fb, 2020-05-20:
closed by #409

fb, 2020-05-20:
Or shall we keep this open until a version containing the change is released?

closed 2020-05-20 by fb