XCSoar/XCSoar #407: Waypoint selection crashes (PPC2003)

GliderManCZ, 2020-05-18:
I have upgraded from v6.8.10 to v6.8.14 for PPC2003 and there is a bug when I choose some waypoint from Waypoint List the XCSoar crashes. I tested the same on Android and it is OK. v6.8.11 is OK, but v6.8.13 has this bug too. I'm using PocketPC HP iPAQ hx4700.

MaxKellermann, 2020-05-22:
I can reproduce this, but only with optimized builds. I have a feeling that this is a compiler bug. We updated the WinCE compiler since 6.8.13.

MaxKellermann, 2020-06-13:
Good news, everyone! The compiler is fine. It's an off-by-one buffer overflow.
https://github.com/XCSoar/XCSoar/blob/6e468340e9dfbfcb56233333878231240221d687/src/Screen/GDI/LargeTextWindow.cpp#L44
This buffer doesn't have enough room for the terminating null byte.
Bug was introduced 9 years ago by commit 7a9c16d6539a7630a5280d0f1a2ed1e2b3dadb19 (@Turbo87)

closed 2020-06-13 by MaxKellermann