XCSoar/XCSoar #408: Flights are logged with year 2000 (started in 2020)

lukasradek, 2020-05-18:
Hello XCSoar people. First of all kudos to you all.

Not changing any settings, once the calendar flipped to 2020, XCSoar started to log flights with wrong date. Time matches, but the whole date is off.

My flight last flight from May 17th 2020 was recorded as October 1st 2000,
my previous flight from May 4th 2020 as September 18th 2000.
So the timespan matches, just the timestamp is off.

Version 6.8.14-Android
Date: May 14 2020

Running Android 5.0 on Samsung Galaxy Note 3 (SM-N9005).

Happy to provide further info. I am a programmer (incl. some basic Android) so should be no problem 😉.

MaxKellermann, 2020-05-18:
This very much sounds like a broken GPS: https://en.wikipedia.org/wiki/GPS_Week_Number_Rollover
Until proven otherwise, there is nothing to fix in XCSoar, because XCSoar isn't causing this.

closed 2020-05-18 by MaxKellermann

lukasradek, 2020-05-18:
Digged some more into that, you are exactly right.
Learned something today. Thank you