XCSoar/XCSoar #410: Vario has wrong size

condoratberlin, 2020-05-18:
With git version 8d5c917a5a the vario is not scaled correctly on linux as seen in landscape mode :
[image]
and Portrait mode:
[image]
Additionally the vario needles and bars are not alinged with the markers.

dernatsch, 2020-05-19:
> 8d5c917a5a
Which branch/tag does this correspond to?
I could not find this as a commit identifier.

condoratberlin, 2020-05-19:
I'll take a look tomorrow.

condoratberlin, 2020-05-20:
Ok, i had some fixes to compile xcsoar on fedora which change the git version. it was simply the latest master version. Does nobody else have this issue?

condoratberlin, 2020-05-22:
I could figure out that the bug was introduced by commit  [65cdfef](https://github.com/XCSoar/XCSoar/commit/65cdfef70eda5bd3b804649e4cae1f499823349f).
Prior to this commit the vario needle position was right and after applying this commit the position is wrong. 
But currently i can't see the direct source of the issue.

condoratberlin, 2020-05-22:
Ok, found the root of this issue! The root is a different offset value. 
In the old version the offset was calculated by a new pixel size, where only the have is used for y-direction.
https://github.com/XCSoar/XCSoar/blob/44c552a1f295b715290650de7dd3c4fbb66f0c04/src/Gauge/GaugeVario.cpp#L711-L712
In the commit all these operations are moved to Geometry struct by.
https://github.com/XCSoar/XCSoar/blob/65cdfef70eda5bd3b804649e4cae1f499823349f/src/Gauge/GaugeVario.cpp#L722
Inside the Geometry struct constructor, the offset is calculated by GetCenter().
https://github.com/XCSoar/XCSoar/blob/65cdfef70eda5bd3b804649e4cae1f499823349f/src/Gauge/GaugeVario.cpp#L45
In GetCenter the value for x is also devided by 2 (which is correct, because the center is asked for)
https://github.com/XCSoar/XCSoar/blob/65cdfef70eda5bd3b804649e4cae1f499823349f/src/Screen/Point.hpp#L127-L128
which cause the wrong scale compared to the old implementation, where the x value was not divided by 2.
I don't fully understand the offset value and therefore logic behind is not very clear to me, but there are two possible solutions for me. 

- The simplest one is to multiply the offset.x value by 2, which is a little bit inefficient, because the value was previously calculated by dividing by 2.
- calculate the values for offset direct in Geometry constructor instead of inside GetCenter and use direct the correct values.

Maybe @MaxKellermann is the best person to suggest the best solution. I can provide a pull request if favoured.

closed 2020-05-22 by MaxKellermann