XCSoar/XCSoar #415: ERA80-Xcsoar

vargazsolt20e, 2020-05-22:
Hello,
I fly with an Era80 with connection to xcsoar. My problem is:
The Era sends the barometric altitude. The problem is that the Xcsoar needs now the standard batometric altitude, but the era sends the QNH corrected altitude. The xcsoar thinks that it is the standard altitud and corrects this for qnh. So it shows a completely wrong height because it is corrected 2x to qnh.
it would be a great driver for era, because now the era-xc combination is useless.. Thank you!
Regards
Zsolt

rawtaz, 2020-05-22:
Related: #389

lordfolken, 2020-05-24:
So both companies (lxnav and lxnavigation) spec the LXWP0 sentence:
Lxnav barometric altitude:
https://github.com/XCSoar/XCSoar/blob/48b435a81c2f25297169367f2b65c0b32c8e1c10/src/Device/Driver/LX/Parser.cpp#L50

Lxnavigation in their doc in #389  say:
True altitude in meters

While we could work around it according to the device initialization, such kind of code is a kludge to maintain. Lets talk to lxnavigation and ask for a firmware update.

vargazsolt20e, 2020-05-24:
Hello, thank you for your message. I contacted lx navigation about the problem and they answered the following:

"we have created a Dataport document for our devices and sent it to the development teams of XC Soar, LK8000, Naviter, etc.
It is now on them to correct the calculations they perform and how they receive this info.

​We're still waiting on the development team from XC soar to give us a reply, as the LK8000 and Naviter teams are already working on it."

lordfolken, 2020-05-24:
All the basics (data exchange, bugs, mc) should work out of the box with the LXNAV driver. 
Its just that ERA device is not outputting what is defined in their own specification.

The LXWP0 sentence definition goes back all the way to filser... so changing it for that one device isn't correct. Btw we had the same issue with LXNAV s7 and they also updated the firmware to correct the output.

vargazsolt20e, 2020-05-24:
I understand. It's a really pitty because I think the Era is a great device. I also paid a lot of money for it, but until the problem is not solved, an Era-Xc combination makes no sense at all. It's desperate ... what a pitty. I hope for a happy ending

lordfolken, 2020-05-24:
I'm talking to them via Whatsapp. Lets see if what they think.

vargazsolt20e, 2020-05-24:
Thank you very much!!

MaxKellermann, 2020-05-24:
Wait, what.
So the LXWP0 sentencen was redefined with a different meaning after a firmware update?

vargazsolt20e, 2020-05-24:
unfortunately i don't know, but the current version of xc for openvario and android definitely has problems with the baro altitude of the last era software.

MaxKellermann, 2020-05-24:
We can't fix something we don't know about. Before we can do anything, we must understand the full extent of the problem. If we now change the thing, it might break all other XCSoar users. Certainly, we shouldn't do that.

vargazsolt20e, 2020-05-24:
Hi Max, of course! I wait a bit, maybe there will be some movement. Otherwise I don't see an alternative. With a K6 mux I could filter the data where the baro data is in and get my Flarm to openvario, but then I lose also the Airspeed, Vario, and Mc data and that doesn't make any sense. Would be a real shame if I had to say goodbye to Era.

lordfolken, 2020-05-24:
@MaxKellermann i think its the same issue as LXNAV had with firmware version 2.0 of the S7 Vario. The LXWP0 sentence outputs baro altitude + qnh correction, when it only should output baro altitude without the qnh correction. I don't wan't to change xcsoar, i want them to change it in firmware. IMO XCSoar should run with the ERA out of the box with the LXNAV (yes i know wrong company) driver.

lordfolken, 2020-05-24:
> unfortunately i don't know, but the current version of xc for openvario and android definitely has problems with the baro altitude of the last era software.
> 
> https://photos.app.goo.gl/PeDuaAvW6Svg6SFK9
that link doesn't seem to work. 

Could you attach a log from the communications? Could you also set things like bugs, mc and ballast while you log?

vargazsolt20e, 2020-05-24:
Ok, yes i do next week, i don't have now a log, but  make one, when i'm on the airfield. Thank you! 

https://www.flickr.com/photos/188597845@N04/albums/72157714440677368

vargazsolt20e, 2020-06-01:
Hello everybody!
Unfortunately after several flights the situation with Era-Xc is really frustrating with the barometric altitude problems.
My question; Wouldn't it be a solution if I connected my flarm directly to the Nmea connection at "A" position and from there I was supplied with GPS data and Baro data and at "B" position, I close the ERA and from ther take the Vario, Airspeed , setting data and ignores the gps and baro (because the second "B" place is)?

rawtaz, 2020-06-01:
What you propose is a workaround, not a solution :) It would be best if the real root cause of the problem could be solved.

vargazsolt20e, 2020-06-01:
Unfortunately I can not help to solve the bug, because my expertise in computers and programming almost ends when I turn on the computer. I am a enduser and I need a pragmatic solution for today, yesterday, yesterday, because at the moment the most important reason why you use a glider software (finalglide calculation, distance to airspaces, etc.) is not working properly, which is not only annoying but under circumstances the wrong data can be dangerous.

rawtaz, 2020-06-01:
That's fine, we understand. But if you could get the log @lordfolken asked for it would be very useful! Sometimes things don't work and in this case it seems we have a hunch why, but we need to be systematic about fixing it.

vargazsolt20e, 2020-06-01:
you're right! I completely forgot to do it.:(
 i will make. thanks!

MaxKellermann, 2020-06-01:
Any further discussion is a pointless waste of time until we have reliable information about the problem from both manufacturers.

vargazsolt20e, 2020-06-13:
> That's fine, we understand. But if you could get the log @lordfolken asked for it would be very useful!
Hello, 
Ok. yesterday I finally got to fly. I made 2x debug for 10min. on my Vertica pna, but I can't find it anywhere on the device.... It just wrote an Igc and any ". BIN" files.
Any idea?
Thank you!
Regards
Zsolt

elgandoz, 2020-06-15:
I'm unsure if this is related, but it could very be.
Yesterday I flew with a brand new ERA80 unit. After pairing just before takeoff, inflight I quickly realised the Barometric Altitude piped to XCSoar was completely off, of tens of thousands feet (AU), while GPS altitude was displaying the correct altitude (same as the ERA and analog barometer).

After analysing the IGC I understood the piped information was numerically in feet but sent as meter. As result, my flight with top height of ~17.000 ft became an hyperbolic ~56.000 ft (which is ~17.000 mt).

I couldn't spend too much time, but I couldn't find any settings for this on XCSoar or ERA.
I'm using the "LXNAV" driver (yes, wrong manufacturer) with "Generic" second Driver, passthrough and both syncs "On".

If it's handy, [here you can see my log ](https://skylines.aero/flights/120061)where you can get the IGC. If you need any other log from the ERA, I will be happy to collect all the information you need in in the upcoming weekends.

fb, 2020-06-15:
@vargazsolt20e @elgandoz: I pretty sure @lordfolken was thinking about a NMEA communication log. If I remember correctly,  this can be obtained from via the XCSoar devices dialog.

> All the basics (data exchange, bugs, mc) should work out of the box with the LXNAV driver.
> Its just that ERA device is not outputting what is defined in their own specification.
>
> The LXWP0 sentence [...]

I agree 100%. *ERA Company* must fix this in their firmware.

Having a workaround in XCSoar is not correct. What will happen if other companys also decide to have their own version of `$LXWP0`?

vargazsolt20e, 2020-06-15:
I understand, but
I can’t force anyone with the gun. The company said their point of view pretty clearly. I have described the problem with the local lx dealer in German-speaking countries. One did not respond, the other is also of the opinion that the xc should make the changes. Unfortunately, as a final user, I can't do anything more and I separate my Era80 from Xc and I switch off the data that contains the Baro data because I have no better idea.
Is a real shame because it could be a great thing.
Thank you anyway!
Reg.
Zsolt

rawtaz, 2020-06-15:
For those of us that don't know german; What reason/motivation does LX give for not fixing an evident bug in their firmware/output? What's their reasoning/rationale for not fixing something that is broken (in the right place - suggesting it should be worked around instead of fixed isn't an option).

vargazsolt20e, 2020-06-15:
That's a good question. I don't understand. Maybe the Zeus devices work with such sentences and don't feel like changing everything ..? I dont know. I can only warn Era users, because at the moment it is really not a good working system with xc.

mlep, 2020-06-15:
@vargazsolt20e Maybe you could add the Era to the lisk of hardware (https://www.xcsoar.org/hardware/), with a warning?
This file needs to be edited: https://github.com/XCSoar/website/blob/master/hardware/index.md

MaxKellermann, 2020-06-15:
> the other is of the opinion that the xc should make the changes.

And my opinion is that XCSoar should not support known-broken peripherals. If we know the ERA80 is buggy, then state clearly that XCSoar is not compatible with ERA80.

(We still don't have reliable information about the problem, do we?)

lordfolken, 2020-06-18:
We really need nmea log of this. All we have is the xcsoar igc file linked above. 

To be fair: 
lxnavigation (or rather the italian reseller)
 - has published a protocol document publicly
 - offered a dev device

Integration issues with a new device are to be expected.

I think the second part of the nmea protocol they published is interesting:
It allows to set tasks / polar data and turn points, radio frequencies.

elgandoz, 2020-06-21:
I just came back from the airfield, thinking to have collected the logs from 2 different phones, one running the XCS latest stable and the other running latest testing. Unfortunately when I checked from my laptop the the NMEA sentences were not in the xcsoar.log file!

This is the procedure I did:
- Enable Bluetooth on my phone and then start XCSoar
- Go on the device dialog, and make sure the ERA was connected (see screenshot 1)
- Check the monitor to see the output (see screenshot 2)
- Close the monitor dialog and press debug: a confirmation appeared that log were being recorded for the next 10 minutes.
- Fly
- After landing I shutdown XCS and copied the xcsoar.log in a different folder.

I realised the sentences were not in the log only after checking the files from a laptop. I also accidentaly overwritten one non-old log while I was transferring it from the second phone.
So..to make sure I'll do it right next time...
Where does xcsoar store the debug log? Is it in a different file than xcsoar.log (or xcsoar-old.log)? Maybe it's still in my device, but I couldn't find anything.
If not, what did I do wrong in the above procedure?

I'm sorry about the mess up. I'll likely be back next weekend.
I'll attach regardless folder with the logs: https://drive.google.com/drive/folders/1Y9LXWxdYBejE1ypmx4jdhKzCmduruhB2

lordfolken, 2020-06-21:
$LXWP0,N,0.0,**2452.1**,-0.00,-0.00,-0.00,-0.00 ,-0.00,-0.00,,301,12.2*70
$GNGGA,052500.50,3608.05473,S,14907.89793,E,1,12,0.66,**776.9,M**,9.8,M,,*6a

So those 2452.1 should be in meters.If its feet it should be about 817m which is much closer to the 776.9 in the GNGGA sentence coming from your GPS, probably a flarm in this case.

So according to the document from LX Navigation in #389 
Section 1.4.3 LXWP0(User <-> LX) describing the sentence structure:

parameter
<is_logger_running> char "Yes" or "No"
<tas> float True airspeed in km/h
**<altitude> float True altitude in meters**

So this is really a device issue and needs to be fixed by the manufacturer.

MaxKellermann, 2020-06-21:
> So this is really a device issue and needs to be fixed by the manufacturer.

WTF. So LX Navigation f*cked up the ERA80 firmware *big time*, by mixing up imperial and metric units.

There is *nothing* XCSoar can and will do. This bug report should be sent to LX Navigation, not to XCSoar.

lordfolken, 2020-06-21:
So my mail to info@lxnavigation.com:
Hello,

We had a user reporting a fault while trying to use the ERA-80 with xcsoar.

He was getting wrong altitudes in the igc log file:

https://skylines.aeroflights/120061

Upon investigation I discovered that the LXWP0 sentence that the ERA is outputting seems to output the altitude in feet instead of meters:

$LXWP0,N,0.0,2452.1,-0.00,-0.00,-0.00,-0.00 ,-0.00,-0.00,,301,12.2*70
$GNGGA,052500.50,3608.05473,S,14907.89793,E,1,12,0.66,776.9,M,9.8,M,,*6a

So those 2452.1 should be in meters.If its feet it should be about 817m which is much closer to the 776.9m in the GNGGA sentence coming from the GPS, probably a flarm in this case.

So according to the document from LX Navigation in #389
Section 1.4.3 LXWP0(User <-> LX) describing the sentence structure:

parameter
<is_logger_running> char "Yes" or "No"
float True airspeed in km/h
float True altitude in meters

So in my opinion this is an implementation error in the device.

LXWP0 has been in the use all the way back to lx4000 and many many devices and software depend on this structure to be meters. So changing it in xcsoar is not an option.

Please note that the output should be barometric altitude.

Could you provide a firmware update, so we could retest this?

Regards,

 - Philipp Wollschlegel

lordfolken, 2020-06-21:
@elgandoz can you find out what the firmware version of your era is?

elgandoz, 2020-06-21:
I took notice and forgot to mention it: 10.6 1.6.70/6381 (2020-05-12)

lordfolken, 2020-06-24:
So they have responded:

Hi Philipp,

thank you for notifying us!

We have discovered this issue and fixed the unit conversion in a later update.
You can find the latest update here: XXXXXX
Regarding the true vs. barometric altitude, we send the true altitude in the LXWP0 sentence. This is in line with our protocol.
XC Soar would like to get the Standard altitude, which you then correct in regards to QNH. 

If this is what you need, you have in the LXWP3 sentence the altitude_offset parameter, which is the difference between true and standard altitude, so you have both pieces of data available and can calculate the info you need.

Kind regards,

(Identifiying information and link removed.) 

So at this point:
 * They are providing a dev device
 * They have published their protocol in written documented form
 * They respond to requests. 

So I will have a look at the driver now.

lordfolken, 2020-06-24:
@elgandoz could you provide me with a mail address. then we can try that firmware update.

vargazsolt20e, 2020-06-24:
Philipp, may I also have the last update from Era? I have version 1.4.40 and the problem with QNH Baro altitude is still stupid.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

lordfolken, 2020-06-24:
MLX_Era_1_6_119_6478.lxu is the firmware filename. I'm not sure wether i'm allowed to share it freely, as it is not available for download from the homepage.

elgandoz, 2020-06-25:
Thanks @lordfolken. Could you please send the email to canberragliding@gmail.com. I will update this issue when done. 
@vargazsolt20e I'm not sure why they're not publishing the firmware update, but we got other updates in the past through the reseller or contacting them directly.

lordfolken, 2020-07-21:
@vargazsolt20e the firmware is now on the homepage.

vargazsolt20e, 2020-07-28:
Hi Pilipp,Yes I know. I've also got beta versions from Luka, but the story with QNH standard height is still a problem. (qnh sent height, again corrected by xc qnh)Reg. Zsolt--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

   
  Reopened #415.

vargazsolt20e, 2020-07-28:
the following sentence stands for xcsoar in the Era revision history;1.6.112/6457 (2020-06-10) • Protocol extended with additional navigation and sensor data as requested by LK8000 and XCSoar developers--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

lordfolken, 2021-04-14:
lx era seems to have recieved a new firmware 2.0: https://downloads.lxnavigation.com/downloads/software_updates/updates/LX_EraRN_R6.pdf 
https://www.lxnavigation.com/support/era/lx-era-80

lordfolken, 2021-04-14:
* Fix  for  TCP  communication  not  working  (problem  with  Mode  TCP/or  UDP  setting)  +LXBC,AHRS default interval increased to 1s

vargazsolt20e, 2021-04-16:
Hallo Philipp, auch mit die neue Update funktioniert nicht deiese Schexxxx... Schade! LgZsolt--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

lordfolken, 2021-04-16:
That's too bad.. i'm really dissapointed. The device is quite nice both in design and features. I think an under-advertised feature, is the can bus thing, where you could hook all the devices together as you wish. Alas the software maturity / overall stability just isn't there.. yet.

vargazsolt20e, 2021-09-20:
>   I tried to implement this driver last year. The issue was that nothing worked as advertised. I couldn't set data rates, request the correct protocol sentences. When you openened something on the menu on the device the nmea sentences would not come constantly.. Sometimes sentences would only arrive on the cable, but not on bluetooth etc. The Protocol is difficult. You need 3 sentences in order to get the correct altitude...
>   The work with the manufacturer was good. They responded to my requests quickly and i got several bugfix firmwares. But the real problem is that i felt like their QA department. I might try again with the 2.0 firmware they released this year.

Hallo zusammen! Leider kein änderung. Die Höhe stimmt  nach wie vor mit aktuellen Era SWUpdate. LgZsolt--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

vargazsolt20e, 2021-11-11:
Hello, everyone. I tried again now with Era and xcsoar with BT connection (all sw up to date) unfortunately the height on the Xc is still not correct.RegardsZsolt--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

vargazsolt20e, 2021-11-14:
with condor driver i have finaly the correct altitude...and airspeed and vario data. only the MC, Bug, Wather setting dont work. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

tmignien, 2022-04-24:
Hello, do we have any update on this please ?

Would it be simply possible to clone the Device driver from LXNAV to LXERA, and to discard the baroaltitude data coming from it ? I know this is not a viable solution, but that could stop the confusion, with a clear warning for everyone that baro altitude is not supported for this device. This in favor of other sources for baro altitude (like the Flarm, or internal if available).

This also to prevent the logging of corrected pressure altitude into the IGC file instead of baro altitude... which breaks the rules.
Then if someone have any success in dealing with LXWP3 data, this new driver could evolve re-enabling the baro-altitude datasource.

Moreover, after some readings, the LX ERA Manual also states that the corrected altitude value does not apply QNH, but maintains the manually defined altitude during device boot-up until switching to the flight mode at take-off. Atre take-off, the LX ERA start to use the QNH value to make corrections. That is to say that XCSoar may perform initial QNH auto calibration before take-off based on fixed and manually entered altitude on the ERA.. and not QNH corrected.

I've just made a analysis on such an IGC record coming from XCSoar connected to a LX ERA unit, I can cleary see the switch in corrections before and after take off, by graphing difference between gps altitude and altitude data sent from the ERA.

MaxKellermann, 2023-05-09:
I guess the bottom line is that the ERA80 behaves in strange undocumented ways, and apparently nobody cares, therefore XCSoar doesn't support this product. In any case, it's not a XCSoar bug, but a bug in the vario.

closed 2023-05-09 by MaxKellermann

Dorianosaur, 2023-09-15:
So thought I'd jump back in on this because I found a workaround, providing you have any other Bluetooth device that provides Baro.  In my case, I have a power mouse FLARM. 

With the FLARM set as Device B and the Era set as Device C, XC-SOAR takes the baro and FLARM data directly from my FLARM, and because FLARM doesn't provide airspeed or vario, XC-SOAR takes those from the ERA. I'm yet to fly with this setup but at my desk, it appears to work.