XCSoar/XCSoar #417: Repository files update availability

dernatsch, 2020-05-22:
On IRC @lordfolken suggested a system to notify about and automatically download files from the repository that are out of date.
This morning I put together a [solution](https://github.com/akafliegdarmstadt/XCSoar/tree/auto-file-update). It checks the update field in the repo vs the time of last change of the files.

Now what I came up with is deeply embedded into the file manager widget, so accessing it programmatically is near impossible.

Another, solution I can think of would be a kind of `RemoteResourceManager`.
An object that would completely manage the providing, downloading and updating of these files centrally. This would allow access from other parts of the program (i.e. updating the files automatically on start if the user wants it). But a lot of changes in the way these files are accessed would be necessary.

Maybe somebody has some input as to what would be the nicer solution here.

MaxKellermann, 2020-05-22:
I think your code is good, and we can merge it. Just a few things need further discussion:
- displaying file update time vs. showing when we updated the file in our list - I think we should tell the user when the file was current (e.g. which iteration of airspace changes this reflects), so he can compare it with official dates
- maybe we shouldn't really work with dates, but with checksums or version numbers? Many people have bad clock values (though rarely on Android), and using the system clock is often not the best idea and not so reliable.
- should this new feature really be merged in the stable branch?

dernatsch, 2020-05-22:
1. Certainly good for the user experience. But I think this would push a bit into the `Manager`(or however a more independent solution would look like) direction because you would need to look up this metadata for all of the locally stored files, in contrast to only probing the file update time.
2. I think the unreliability of the local clock is a good point.
A checksum would be nice because it can be calculated directly from the data, XCSoar could compute it on the fly. A version number would have to be managed/incremented somehow and would need a place to be stored (a separate file, the filename, the content itself?). In the end its a question of compute vs store I think.
3. I can rebase it of course.

dernatsch, 2020-06-06:
Two things I'm currently wondering about for sha256:

1. How to get the checksum the download generates. What comes to mind is passing it by value in `OnDownloadComplete` and adding the LocalRepository to the listeners, **or** making the DownloadManager save the hashes of the files instead of the LocalRepository (not a huge fan because of the DownloadManager being global).

2. How to persist the hashes. Recalculating the hashes could take a long time so saving them in a file could be a simple fix. But how? A CSV file? Or would it be viable to rehash the files concurrently and only providing the 'out of date'-functionality when it's done.

lordfolken, 2020-06-06:
The issue will always be that we do not control the source. So what may happen is that the download doesn't match the checksum in the repository file, falsely showing that there is an update.
Since the aim is to show an update when the repository updates, i'd currently save the checksum to a csv file, with the filename update date and checksum. If any of those do not match, xcsoar shows an update.
If we'd want a reliable checksum, we would need to download all the files in the repo and  devise a cache invalidation mechanism.

dernatsch, 2020-06-06:
So you're saying it should just save the checksums it gets from the repo file instead of computing them locally?

lordfolken, 2020-06-07:
At least for the time being. The repo also needs some intelligence upgrade. I implemented good enough[tm]

lordfolken, 2022-05-20:
So there is currently an update function in the file manager, that allows you to update resources that have updates. 
What is missing is a way to notify the users of updates, without requiring them to enter the file manager dialog.

lordfolken, 2026-09-10:
Related architecture: #3086 / apply-in-session: #3092.

Auto-download (or “update available”) only writes a newer file to disk. That is the same gap as File Manager “update all” (#1115): the running session keeps the old map/CUP/OpenAir until something reloads the **configured** path.

Do not add a `RemoteResourceManager` (or similar) for this. File Manager already owns the download; #3092 is the missing apply step. Auto-update UI can land later on top of that.

closed 2026-10-21 by lordfolken