XCSoar/XCSoar #424: XCSoar 7.0 triggers asserts in debug when position isn't acquired and user changes MC, Bugs

vyzaq, 2020-06-03:
XCSoar versions having and not having the problem
-------------------------------------------------
I found that bug on master branch (commit 9f4f51daeb17b9d4ad5ea709f39fac5b1106b3c8) during development of the driver for LX Navigation devices.

Expected behavior
-----------------
XCSoar should work as usual when location isn't avaialble.

Actual behavior
---------------
Multiple asserts triggered, according to the code invalid location is used for Glider computer calcualtions.

Steps to reproduce the behavior
-------------------------------
1. Start the application
2. don't wait for GPS location and start to change MC value (usually +2 or +3 m/s and back to zero)
3. Asserts triggered GlideComputerAirData.cpp: GlideComputerAirData::CruiseGR() method, 
Sometimes asserts triggered in Device/Parser.cpp: NMEAParser::GGA() method when GGA sentence contains data that GPS hasn't acquire position.

Do you have any idea what may have caused this?
-----------------------------------------------
Listed above methods don't check input parameters

Do you have an idea how to solve the issue?
-------------------------------------------
There is a patch I created but I'm not sure that the fix correct.
```
diff --git a/src/Computer/GlideComputerAirData.cpp b/src/Computer/GlideComputerAirData.cpp
index b1a0f58189..39120464eb 100644
--- a/src/Computer/GlideComputerAirData.cpp
+++ b/src/Computer/GlideComputerAirData.cpp
@@ -212,6 +212,9 @@ GlideComputerAirData::GR(const MoreData &basic, const FlyingState &flying,
 inline void
 GlideComputerAirData::CruiseGR(const MoreData &basic, DerivedInfo &calculated)
 {
+  if(!basic.location.IsValid() || !calculated.cruise_start_location.IsValid())
+    return;
+
   if (!calculated.circling && basic.NavAltitudeAvailable()) {
     if (calculated.cruise_start_time < 0) {
       calculated.cruise_start_location = basic.location;
diff --git a/src/Device/Parser.cpp b/src/Device/Parser.cpp
index aba56faea5..a3eb9917c2 100644
--- a/src/Device/Parser.cpp
+++ b/src/Device/Parser.cpp
@@ -654,7 +654,7 @@ NMEAParser::GGA(NMEAInputLine &line, NMEAInfo &info)
     // If the separation doesn't appear in the sentence,
     // we can assume the GPS unit is giving ellipsoid height
     //
-    if (use_geoid) {
+    if (info.location.IsValid() && use_geoid) {
       // JMW TODO really need to know the actual device..
       geoid_separation = EGM96::LookupSeparation(info.location);
       info.gps_altitude -= geoid_separation;
```

MaxKellermann, 2020-06-03:
None of this is correct.
`basic.location.IsValid()` can't be used because the location is an uninitialized variable if there is no GPS fix yet (which is different from `!IsValid()`); check `location_available` instead.
Checking `calculated.cruise_start_location.IsValid()` is also incorrect, because that variable will be calculated by that method. If you never allow it to enter, it will never be calculated, which breaks the whole feature.

vyzaq, 2020-06-03:
ok, got your point, will change the code and try how it works and create pull request later. But Is it the correct place for "if" checks here? Asking because I'm not so familiar with that part of the project.

MaxKellermann, 2020-06-03:
Yes, that looks right. Every function which accesses a variable must be confident that the variable was initialized. Functions can do it by documenting this as a requirement for the API (and if so, there should be an `assert()` validating this in debug builds), or they must implement their own checks.

closed 2020-06-28 by MaxKellermann