XCSoar/XCSoar #425: add two more additional airspace fields

mcatgithub, 2020-06-03:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

  
What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

rawtaz, 2020-06-03:
I have felt the need for more airspace file selections too, I second this feature request! 👍

lordfolken, 2020-06-03:
A file selection dialog where you can tick or untick N ammount of files would be interesting. The files would then get merged into one and loaded.

rawtaz, 2020-06-04:
> A file selection dialog where you can tick or untick N ammount of files would be interesting. The files would then get merged into one and loaded.

As long as the files are merged every time they are referenced, because if one of the files are updated it's not practical that the profiles referencing that file has to reconfigure or otherwise trigger a re-merge of the data.

For example, if you have selected three files in your profile, and one of them is updated, then after a restart of XCSoar the updated file must be used without the user having to reconfigure/re-trigger the merging.

Buzz33, 2020-06-24:
I also support this request.

Flying in the Alps I am currently using Germany, Austria, Slovenia, Italy, Swizerland (Mil On or Mil Off) and France, plus a separate self made file for special Airspace like the Tauern National Park. Thats 7 in total. Currently I have to manually merge these 7 files  into two files.  

So I would be very happy to have all these 7 files in separate slots.

closed 2025-08-22 by lordfolken