XCSoar/XCSoar #426: point & click on the map, very difficult when it's bumpy

bomilkar, 2020-06-04:
XCSoar version
--------------

7.0 and 6.8

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

When pointing on the map to select an airport or any other item, the hit range is so small, that I can hardly get the airport on the list "Map elements at this location". I have to be spot on and that can be nearly impossible in flight.
When using a touch screen the hit radius is sufficiently wide. That works. But when using a mouse/keyboard/stick_remote it is nearly impossible to get the desired map element.

So I suggest to make the hit radius wider for devices having no touch screen.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

It'll be easier to point at the map and get the desired airport on the list "Map elements at this location".
It'll affect a single line in src/Screen/Layout.cpp line 134
instead of `hit_radius = PtScale(HasTouchScreen() ? 24 : 6);`
make it `hit_radius = PtScale(24);`

kedder, 2020-06-04:
This is very much relevant to OpenVario users, who have got a mouse pointer movement controls on the stick. 

However for that case I'd suggest making a openvario-specific patch that always sets touch screen to be available (regardless of actual hardware availability). This will address this issue and also will make all UI buttons larger (with text actually fitting into button area).

Or, some kind of configuration/command line option that forces `HasTouchScreen()` to return true would be nice as well.

bomilkar, 2020-06-04:
Good point, we might add `IsOpenVario()` to Asset.hpp and evaluate for `HasTouchScreen()` .

MaxKellermann, 2020-06-04:
Why would OpenVario be special? We need a different, generic way of determining "we need a bigger hit radius". Obviously, touch screens need a larger radius than a mouse, but what exactly makes the OpenVario different that it needs a larger radius?
Does XCSoar miscalculate the physical screen area? Is the physical screen size correctly determined on the OpenVario? Or does an OpenVario really need a bigger *physical* hit radius? If yes, then why?

Plantain, 2020-06-04:
Most OpenVarios don't have a touchscreen but rather use a 'mouse' (like a
Thinkpad nub on the stick), not a touch screen - and using that mouse in
turbulence is the difficulty.

rawtaz, 2020-06-04:
I fail to understand how. You have your hand on the stick, and operate the mouse pointer with your thumb, how is that susceptible to turbulence?

kedder, 2020-06-04:
I think it doesn't have to do with turbulence, it is just somewhat harder to place cursor precisely using arrow keys. Especially in situation when you can't concentrate your attention on the arrow for a long time.

I don't think the solution should be specific to Openvario, I just used it as example and to explain where the issue comes from. I also found that `HasTouchScreen()` controls the size of some buttons on screen, and XCSoar looks better with larger buttons (especially when font size is bigger than 110%).  

This can be solved by applying custom patch for xcsoar in openvario build, without involving this repo at all. But if someone would suggest a generic solution that will fit into scope of XCSoar, that would even better.

MaxKellermann, 2020-06-04:
So if this is about the type of pointing device, maybe we need a third class if pointing device?
Buttons (and scroll bars, ...) are larger on touch screens because they too need a bigger hit radius for fat fingers tapping on them. This is mostly controlled by `maximum_control_height`:
https://github.com/XCSoar/XCSoar/blob/f96bbd1db2b0628728a2234afee4f75c99aac1f3/src/Screen/Layout.cpp#L124-L131

bomilkar, 2020-06-04:
Part of the problem: the map moves while the pilot is trying to move the cursor to the object and anticipate the right moment for hitting the button.
A hit_radius of 6 pixels is definitely too small such a maneuver.
What kind of pointing devices do we have in the gliders predominantly? I guess it's touch screen, nub, or 2 rotary encoders. No mouse, no touch pad or alike.
I don't care too much about the workstation at home where you probably have a mouse or a touch pad and the map may not be moving. Totally different scenario.

MaxKellermann, 2020-06-04:
>  A hit_radius of 6 pixels is definitely too small for such a maneuver.

It's not 6 pixels. It's 6 "points" (one "point" being a 1/72 inch) = 1/12 of an inch. It's `PtScale(6)` pixels.
https://github.com/XCSoar/XCSoar/blob/f96bbd1db2b0628728a2234afee4f75c99aac1f3/src/Screen/Layout.hpp#L56-L60

MaxKellermann, 2020-06-04:
Again: did you OpenVario people make sure XCSoar knows the right dpi value of the screen?

kedder, 2020-06-04:
@MaxKellermann that's a good question. The real value is around 134dpi (varies slightly depending on display model), but I'm not sure how XCSoar calculates it (when "Display Resolution" setting is "Automatic"). Where does it get the physical size of the display from? OpenVario running in KMS mode, i.e. not under X.org.

kedder, 2020-06-04:
It actually feels like the autodetected value is much lower than actual. When I start XCSoar with explicit dpi setting (`xcsoar -dpi=134`) the airspace borders are noticibly thicker. I suspect the automatic dpi is 96 (good old default from 1990s) or something like that.

MaxKellermann, 2020-06-05:
dpi detection is unimplemented and thus hard-coded for the KMS/GBM code. See https://github.com/XCSoar/XCSoar/blob/f96bbd1db2b0628728a2234afee4f75c99aac1f3/src/Hardware/DisplayDPI.cpp#L106-L108
Somebody needs to implement that.
Does the kernel really auto-detect the density of OpenVario's connected display? Does the display allow to identify its densitiy?

bomilkar, 2020-06-05:
I'm using the 7" Texim display. According to the datasheet it has 133.3 dpi in X and Y.
I have not yet seen a method to read this directly from the display. I'm still checking with the manufacturer.
Hard coding it in DisplayDPI.cpp for every possible display type is a mess.
For OpenVario I suggest we keep display characteristics in some config file (unless the display can tell us what it is) and provide these values (resolution and pixel density) at the commend line.

However, there are 2 issues which I'd like to raise for discussion:
1. HasTouchScreen() is not a good surrogate for "fuzzy pointing device". Some touch screens might have better pointing characteristics then nub devices and vice versa. Currently HasTouchScreen() determines the hit_radius. If it thinks it has a touch screen the hit radius is 4x the normal size regardless of the quality of the pointing device. And HasTouchScreen() is simply linked to the TARGET. I think the criteria for hit_radius should be smarter.

2. hitting a moving target is difficult and the hit_radius can only alleviate to some extend. Why don't we implement the following: while the mouse cursor is moving over the map area, we suspend the map movements. So the target is no longer moving. Suspending the map movement should extend for a couple of seconds after the mouse pointer has stopped to give the pilot the chance to press the select button.

kedder, 2020-06-05:
> Does the kernel really auto-detect the density of OpenVario's connected display? 

Technically the kernel knows the physical size of the display from the device tree description ([example](https://github.com/Openvario/meta-openvario/blob/226c2e53a1af15fbecaab8cbf823e8f035f4121a/recipes-kernel/linux/linux-mainline/openvario-7-CH070.dts#L3-L6)), but I'm not sure if (and how) this is exposed to userspace. While it would be nice to be able to determine dpi automatically, it would probably not solve the problem described here. The hit radius would grow 30% with the correct dpi at best.
 
> while the mouse cursor is moving over the map area, we suspend the map movements. 

This can backfire pretty badly if you have a jiggly input device (say, trackball). You might never see a  map moving that way.

bomilkar, 2020-06-06:
> This can backfire pretty badly if you have a jiggly input device (say, trackball). You might never see a map moving that way.

Good point. I didn't think of jiggly pointing devices.
So what if instead of suspending the map movement for a while let the mouse cursor move with the map while it is over the map area? The pilot uses the pointing device (mouse) to move the cursor relative to the map instead of the display. Over time the mouse cursor moves to the edge of the map if it was left off over the map area.
If you think this is an esoteric idea: I'm trying to improve the hit probability wile keeping the hit radius as small as possible. Because there is a downside of increasing the hit radius: the list "Map elements at this location" includes an increasing amount of irrelevant objects.

MaxKellermann, 2020-06-08:
dpi detection for KMS has landed in v6.8.x: https://github.com/XCSoar/XCSoar/commit/df4d0b372fa97cf725ad503489647d60574502ea
Please check if this detects the OpenVario display size correctly.

lordfolken, 2021-06-09:
@kedder can we close this?

lordfolken, 2021-07-05:
@kedder @bomilkar is this now solved?

bomilkar, 2021-07-05:
I checked yesterday and in my opinion it's not solved.
It still uses `HasTouchScreen()` to determine the hit radius. But not every device with the need for a wider hit radius has a touch screen. This criteria is just not right.
But it's a simple patch to src/Screen/Layout.cpp to widen the hit radius if needed.
Maybe in the long run we need to introduce (yet another) parameter in the prf.

lordfolken, 2022-06-08:
I think the hit_radius should depend on the method with which the cursor movement occurred:
 - If its been via touch => 24
 - If the mouse cursor was used => 6
 - If the screen has been moved by the cursor keys 1/4th the width of the screen (this due to the implementation here):
 https://github.com/XCSoar/XCSoar/blob/a64b0fbe64742f269bbd325bd1c6fa976ef1265f/src/Input/InputEventsMap.cpp#L165-L169

Especially in the last case, its not really possible to select all way points on the map.

elgandoz, 2022-06-09:
I would like to propose 2 small adjustment concerning this issue:

1. While panning, the crosshair in the center on hight DPI screen is pretty much invisible: is it possible to increase the size/thickness (or any other way to make it more visible with the screen at cockpit distance)?
2. The order of the elements in the list "What's here?" seems to be a mix of alphabetical & type: would it be possible to make it ordered by proximity (from the crosshair) and type (first waypoints, then cylinders and airspaces)?
Given that the first one is already selected, it would allow less clicks. As it is now, even if I am spot on, the first item is often not the one I want.

lordfolken, 2022-06-10:
> 1. While panning, the crosshair in the center is pretty much invisible, on hight DPI screen: is it possible to increase the size/thickness, or any other way to make it more visible at cockpit distance?

You are correct.The DPI isn't applied here. The look specifies one pixel width, not using any of XCSoar's scaling functions. 
https://github.com/XCSoar/XCSoar/blame/a64b0fbe64742f269bbd325bd1c6fa976ef1265f/src/Look/OverlayLook.cpp#L34