XCSoar/XCSoar #429: macOS: running XCSoar shows black window

sandrinr, 2020-06-05:
I am trying to the get XCSoar to run again natively on macOS. It used to work but now it does not anymore (for me). I managed to get it to compile (with these build fixes: https://github.com/sandrinr/XCSoar/tree/fix-macos) but when running it, it will just show a blank (black) window.

XCSoar versions having and not having the problem
-------------------------------------------------

Having: XCSoar master
Not having: don't know

Expected behavior
-----------------

XCSoar starts and shows fly/simulate screen.

Actual behavior
---------------

XCSoar shows blank black screen. Input seems to work, I can guess and press where the "Fly" and "Simulate" button is and press it. On the console I can see the program advancing but the window remains black.

Console output when invoking with `--fly` argument:
```
~/Projects/XCSoar (master)*$ output/OSX64/bin/xcsoar --fly
Free storage 66441953280 KB
Starting XCSoar XCSoar v7.0_preview15-macOS~git#e6d420eac1
Display dpi=230,230
OpenGL: npot=1 fbo=1 stencil=0x84f9
Initialise fonts
Startup dialog
Loading profiles
Loaded profile from /Users/raffaele/XCSoarData/default.prf
Loading language file
Loading input events file
OpenTerrain
Loading Topography File...
ReadWaypoints
Failed to read waypoint file: /Users/raffaele/XCSoarData/user.cup
Failed to open 'airfields.txt' from ZIP file
Start at terrain center
RASP load
No rasp data file: Failed to open ZIP archive /Users/raffaele/XCSoarData/xcsoar-rasp.dat
ReadAirspace
Failed to parse airspace file: airspace.txt
Failed to open 'airspace.txt' from ZIP file
Register serial devices
Opening device Built-in GPS & sensors
ProgramStarted
Free storage 66441952256 KB
cannot open /Users/raffaele/XCSoarData/lua/init.lua: No such file or directory
```

Steps to reproduce the behavior
-------------------------------

1. Compile on macOS using my branch with build fixes https://github.com/sandrinr/XCSoar/tree/fix-macos
1. Run run XCSoar: `output/OSX64/bin/xcsoar`

Do you have any idea what may have caused this?
-----------------------------------------------

My guess is that either SDL or OpenGL is not initialized/configured correctly for it to work properly on macOS. However, I don't really have SDL or OpenGL experience.

Do you have an idea how to solve the issue?
-------------------------------------------

Somebody with experience in XCSoar output backends might chime in and point me into a fruitful direction.

MaxKellermann, 2020-06-06:
I tried a very simple SDL/OpenGL/Shaders demo program (https://lazyfoo.net/tutorials/SDL/51_SDL_and_modern_opengl/index.php) and it also shows a black window on macOS.
I heard that Apple has deprecated OpenGL in favor of their own, proprietary rendering API, and OpenGL is now a pure software renderer (even if your machine has a OpenGL-capable GPU). That's sort of proprietary vendor lock-in is obviously a bad thing, because XCSoar will never migrate to that API.
If this OpenGL shader problems has to do with this political decision, then XCSoar will need to drop macOS support completely as a consequence of this issue report.

lordfolken, 2020-06-06:
Vulkan support is available for linux/windows/android and 3rd party for ios/osx. Android 7.0 onwards only though.

MaxKellermann, 2020-06-06:
... but who writes a new `Canvas` implementation for Vulkan?
We have one escape plan for macOS, though: the software renderer (which I developed for the Kobo).

sandrinr, 2020-06-07:
Sad to read about macOS's deprecation of OpenGL and the lack of a real cross platform alternative which also works natively on macOS.

Following @MaxKellermann's point about software rendering, after some minor changes to the source (enabling compilation in the thirdparty section for macOS, fixing font search path for macOS) I managed to build XCSoar with the following build arguments:
```
gmake TARGET=OSX64 OPENGL=n ENABLE_SDL=y FREETYPE=y APPKIT=n
```
With that XCSoar works using software rendering :-). Have not seen any performance issues on my MBP (but I also did not expect any since this stuff also works on the Kobo).

However there are some drawing issues:

Dialog headers:
[image]

Startup screen:
[image]

Then, the main map seems to wiggle under no input:
[image]

With some pointers I might be able to address these.

MaxKellermann, 2020-06-07:
Similar to https://github.com/XCSoar/XCSoar/issues/379 ?

sandrinr, 2021-05-24:
I will close this issue.

There is no change here. However, using Docker to build and run XCSoar does work out of the box on macOS. This is not ideal as a development environment, but unless someone wants to do some serious XCSoar development on macOS, i guess, there is no need for more for the time being.

Feel free to reopen if somebody wants to invest here.

closed 2021-05-24 by sandrinr