XCSoar/XCSoar #430: sodium/crypto_hash_sha256.h: No such file or directory

bomilkar, 2020-06-06:
XCSoar versions having and not having the problem
-------------------------------------------------
7.0

Expected behavior
-----------------

Following the developers manual chapter 2 (Compiling XCSoar) should yield a working binary for the given TARGET, say UNIX64.

Actual behavior
---------------

Compilation terminates because one (or more) files missing.

  CXX     output/UNIX64/dbg/src/Net/HTTP/ToBuffer.o
In file included from src/Net/HTTP/ToFile.cpp:29:
./src/Crypto/SHA256.hxx:35:10: fatal error: sodium/crypto_hash_sha256.h: No such file or directory
 #include <sodium/crypto_hash_sha256.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Steps to reproduce the behavior
-------------------------------

git clone git://github.com/XCSoar/XCSoar
cd XCSoar/
git submodule init
git submodule update
git pull
make TARGET=UNIX64 -j4

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2020-06-06:
Which libsodium version?

bomilkar, 2020-06-06:
I don't know which version is needed.
But the requirement must have been added in past few weeks.

MaxKellermann, 2020-06-06:
Clarifying my question:
Which libsodium version did you install?

bomilkar, 2020-06-06:
I didn't not explicitly install any libsodium. Never had to. And the Developers' Manual doesn't say anything about it.

MaxKellermann, 2020-06-06:
Oh yes it does: https://github.com/XCSoar/XCSoar/blob/48cce1996ce5ce66dcd892f6f791f86eae0ff845/doc/manual/en/compiling.tex#L95
This bug report is bogus.

closed 2020-06-06 by MaxKellermann

bomilkar, 2020-06-06:
Is it missing form `.git/config` ??

MaxKellermann, 2020-06-08:
No. This question doesn't make sense.