XCSoar/XCSoar #433: Internal storage is used instead of SDCard on Moto G LTE

szpak, 2020-06-08:
*I initially wanted to discuss this problem on the forum, but it seems to be down at the moment (403 Forbidden)*

XCSoar versions having and not having the problem
-------------------------------------------------

XCSoar 6.8 for Android from Google Play (also XCSoar v7.0_preview15-Android).
Android 5.1, Moto G LTE (1st gen - XT1039)

It's the first time I install XCSoar on any Android, probably it is a permanent problem.

Expected behavior
-----------------

XCSoarData is placed on an existing SDCard.

Actual behavior
---------------

The internal storage is used instead of SDCard (which is problematic as on this phone it is very limited).

Steps to reproduce the behavior
-------------------------------

1. Start XCSoar on Moto G LTE.

Do you have any idea what may have caused this?
-----------------------------------------------

It took a look at the related source code and in [LocalPath.cpp](https://github.com/XCSoar/XCSoar/blob/6d1946c17eddd2baad08d5700bbacf98fa5162b3/src/LocalPath.cpp#L319-L343) there are the `getExternalStoragePublicDirectory` and `getExternalStorageDirectory` method used. I don't know what returns those method on this Moto G, however, there are:
```
/storage/emulated/0   <- internal storage
/storage/sdcard0      <- internal storage
/storage/sdcard1      <- real SDCard
/storage/usbdisk
```
Most likely `/storage/sdcard0` is used and the directory is created in the internal storage space.

Do you have an idea how to solve the issue?
-------------------------------------------

There could be another "hack" added to detect that situation :-/ (I'm not sure if it is a problem with Android 5.1 or Moto G itself).

Extra comments
-------------------------------------------

I installed `XCSoar v7.0_preview15-Android` to see the debug log statements, but they seem to be disabled in that version as well. Is there a version available with debug logging enabled (or it can be enabled on demand)?

Maybe there is any workaround to force XCSoar to use a SDCard? I was planning to use XCSoar on a glider camp starting this Thursday :-/.

lordfolken, 2020-06-08:
The storage path in use is displayed in the settings->site path dialog at the top.

szpak, 2020-06-08:
Yes, `/storage/emulated/0/XCSoarDate` is used.

lordfolken, 2020-06-09:
XCSoar uses the external storage permission and uses what is returned to it. 

The moto g seems to feature a method of moving applicaitons to the sdcard1:
https://support.motorola.com/in/en/solution/MS105695

"The “On SD card” tab, under Settings ->Apps displays the list of applications which can be moved to the card, as well as these already installed on the card.

You can move each application back and forth through each “app info” settings view (“move to SD card”). Please wait until each move has completed before exiting the App info panel."

szpak, 2020-06-09:
I've tried it, but it seems to be only for data kept by applications in the native way. XCSoar only fetches the remote directory from the system and put files there on its own (without using any Android API). As a result it seems to ignore that change.

Some solution could be to (optionally) keep the path to the data directory in the Android configuration natively and use it (if configured). An user should be able to change it manually in the application UI and point the path which is best. The fallback for the new installations would be as it is now. WDYT?

Plantain, 2020-06-09:
Can you make symlinks on Android?

szpak, 2020-06-09:
I was thinking about that, but unfortunately from my findings it is limited to the rooted devices only.

lordfolken, 2021-06-09:
@szpak can you try again with 7.8? Max fixed a lot of storage related issues.

szpak, 2021-09-26:
@lordfolken Sorry for delay. I've been finally able to give it a try again. I used "Version 7.15 (136) - Added on 2021-09-01" from F-Droid, will the clear configuration. However, by default, it is still `/storage/emulated/0/XCSoarData` (probbly still due to a fact that in that Moto G `/storage/sdcard0` links to the internal storage...). Should I be able to change it somehow manually in 7.8+?

lordfolken, 2022-09-05:
I think this is now anyway a thing of the past. With android scoped storage only the application directory is available.

closed 2022-09-05 by lordfolken