XCSoar/XCSoar #436: XCSOAR 6.8.13 on Kobo glo hd

Flyingbul, 2020-06-11:
Hello! 
The 6.8.13 versionis unstable with **Kobo glo hd**
First installation: seems to work (map appears, GPS ok) but if you play with xcsoar, it becomes unstable. Then, it does'nt work anymore and go back direct to home page. Restart: When charging maps and wp, the map appears briefly then  turn of and return to xcsoar home page. No improvment when reseting, rebooting or so.
Problem apears with or without gps device (external Glonass ublox7), enabling or desabling USB.
New installation of maps: same pattern...
Re-installing 6.8.13: no change
Maps loaded: French Alps, Ful France, standard or HD goes the same way.
Installing 6.8.11 version: it works well.
Best regards
Franck

MaxKellermann, 2020-06-13:
You're using an old version.

closed 2020-06-13 by MaxKellermann

Flyingbul, 2020-06-13:
6.8.13 is a old version ?

MaxKellermann, 2020-06-13:
Try this one: https://download.xcsoar.org/releases/6.8.14/KOBO/KoboRoot.tgz
I released 6.8.14 a month ago, but nobody cared to update the website...

Flyingbul, 2020-06-13:
Thank you very much!