XCSoar/XCSoar #441: Error building with TARGET=KOBO with g++-6

cefn, 2020-07-02:
Build checked out from 6.8.x branch

Expected behavior
-----------------

After following the steps in the Developer manual, installing suitable build tools and libraries, and apparently successfully running `make TARGET=KOBO TCSUFFIX=-6 kobo-libs` the line `make TARGET=KOBO TCSUFFIX=-6` should build a binary (suitable for later bundling into a KoboRoot.tgz).

Actual behavior
---------------

g++ dies with errors which may be to do with changing permissiveness between g++-5 and g++-6 as per the attached file. 

I don't routinely develop with C or C++ so I am unfamiliar with the detail of the errors, but things like the following seem like warnings which may not have been treated as errors with g++-5 - e.g. the code is fine but I need to know how to compile it in a way which permits these unsafe issues as warnings.

```
error: constexpr call flows off the end of the function
error: no matching function for call to ‘std::pair<double, double>::pair(double, double)’
error: ‘value’ is not a member of ‘std::__and_<std::is_constructible<double, double&&>, std::is_constructible<double, double&&> >
```

[compile_errors.txt](https://github.com/XCSoar/XCSoar/files/4861365/compile_errors.txt)

Steps to reproduce the behavior
-------------------------------

I am attempting to build the project from within a Crostini 'Debian Stretch' on my Chromebook, but for build purposes I believe this to be identical to Debian Stretch. 

Do you have any idea what may have caused this?
-----------------------------------------------

Possibly I can't use just any ARM toolchain and XCSoar is dependent specifically on `g++-5`, but I understood that any suitable ARM toolchain from the g++ series >= 5 would do.

Do you have an idea how to solve the issue?
-------------------------------------------

Try to find some way to install g++-5, try to build from master instead, set an `fpermissive` flag somewhere, follow steps to use a virtualised build environment which includes its own compliant versions of everything (e.g. docker).

MaxKellermann, 2020-07-02:
You need at least GCC 8 to build Kobo.

cefn, 2020-07-02:
Thanks, Max. Is there an alternative set of instructions I should be looking at? 

I was following the Developer Manual http://download.xcsoar.org/releases/6.8.15/XCSoar-developer-manual.pdf which points to g++-5 in the build instructions, hence my mistakes above. In my context sounds like I should try to switch to a Debian Buster Crostini VM so I can get hold of https://packages.debian.org/buster/g++-8 and run with TCSUFFIX=-8 instead.

MaxKellermann, 2020-07-03:
The manual is outdated.

cefn, 2020-07-04:
Thanks, Max. I've had some success with a Dockerfile based on the Docker hub `gcc:8` image.

Unfortunately, while I can build version 7.0 (which has render glitches on my Kobo device) and can build the kobo-libs for 6.8.15, like... I run the general make target like `sudo docker run --rm -v "$PWD"/../XCSoar:/usr/src/XCSoar -w /usr/src/XCSoar xcsoar-para make TARGET=KOBO`

cefn, 2020-07-05:
Closing in favour of https://github.com/XCSoar/XCSoar/issues/444

closed 2020-07-05 by cefn