XCSoar/XCSoar #442: KoboGlo v7 Preview build 15 render glitches

cefn, 2020-07-02:
XCSoar versions having and not having the problem
-------------------------------------------------

KoboRoot 7.0_preview15.tgz running on KoboGlo

Expected behavior
-----------------

Preview build is able to render the map and animate it as it moves according to a replayed log file.

Actual behavior
---------------

As you can see at...

https://photos.app.goo.gl/9BKBs2fJhCQNRZHe6

...the version 7 preview build 15 for Kobo shows some unusual rendering glitches, making it unusable.

The same file and the same replay worked in 6.8.15

Steps to reproduce the behavior
-------------------------------

Downloaded and installed the preview, ran a replay against my normal profile from 6.8.15.

Do you have any idea what may have caused this?
-----------------------------------------------

Are version 6 profiles incompatible with version 7?

Do you have an idea how to solve the issue?
-------------------------------------------

Possibly reset my profile completely. Unfortunately I have now reverted to 6.8.15 so hard to test this theory right now.

cefn, 2020-07-05:
Is the Kobo Glo HD is just not powerful enough to fully render the 1448×1072 screen pixels with all the features of 7.0?

If I want to continue to benefit from XCSoar on a daylight-readable eInk screen, would there be an advantage to buying a Kobo Glo (not HD) which has the same CPU power but only 1024×758 pixels to draw? Not sure whether this has a big performance impact.

There's a GPS adapted 1024×758 Kobo Glo available for me to buy here in the UK if a device with fewer pixels would help me to follow ongoing development of XCSoar in v7.0.

lordfolken, 2021-07-04:
Can you try recent builds like this one? http://max.kellermann.name/download/xcsoar/kobo/KoboRoot.tgz

MaxKellermann, 2021-08-23:
No reply

closed 2021-08-23 by MaxKellermann