XCSoar/XCSoar #444: Build failure of stable tag v6.8.15 with reference docker build container

cefn, 2020-07-05:
XCSoar versions having and not having the problem
-------------------------------------------------

The stable v6.8.15 fails, but I can successfully build v7.0_preview15 with the same build environment.

Expected behavior
-----------------

After `make clean` I would expect running the build steps against tag v6.8.15 within the docker build container should create a working KoboRoot.tgz 

This would be equivalent to the stable build distributed from https://www.xcsoar.org/download/ at the time of writing, and help me make practical progress creating and proving software patches given v7.0 is currently [unstable for Kobo Glo HD](https://github.com/XCSoar/XCSoar/issues/442).

Actual behavior
---------------

The build fails as shown below...

```
root@0fbebc07bbcf:/opt/xcsoar# make TARGET=KOBO DEBUG=n USE_CCACHE=y libs
build/libnet.mk:37: *** library not found: libcurl.  Stop.
```

Steps to reproduce the behavior
-------------------------------

I built the docker container following https://github.com/XCSoar/XCSoar/tree/master/ide/docker
then attempted to run the following build steps (the first fails).

```
make TARGET=KOBO DEBUG=n USE_CCACHE=y libs
make TARGET=KOBO DEBUG=n USE_CCACHE=y
make TARGET=KOBO DEBUG=n USE_CCACHE=y output/KOBO/KoboRoot.tgz
```

Do you have any idea what may have caused this?
-----------------------------------------------

I understood the `libcurl` dependency was met by libcurl4-openssl-dev as per https://github.com/XCSoar/XCSoar/blob/2a0940594c4cf035d8a77bf58853bbae49917c9c/ide/provisioning/install-debian-packages.sh#L39  

I have proven that package is in fact installed within the docker container.  

```
root@0fbebc07bbcf:/opt/xcsoar# dpkg --list | grep -i curl
ii  libcurl3-gnutls:amd64                7.64.0-4+deb10u1             amd64        easy-to-use client-side URL transfer library (GnuTLS flavour)
ii  libcurl4:amd64                       7.64.0-4+deb10u1             amd64        easy-to-use client-side URL transfer library (OpenSSL flavour)
ii  libcurl4-openssl-dev:amd64           7.64.0-4+deb10u1             amd64        development files and documentation for libcurl (OpenSSL flavour)
```
My guess is that it's not properly pointed to within the v6.8.15 build config.

Do you have an idea how to solve the issue?
-------------------------------------------

Perhaps some additional flag is needed to point the build to the place that libcurl4-openssl-dev provides the file. 

`dpkg` reports the location of files provided by the package as follows...

```
root@0fbebc07bbcf:/opt/xcsoar# dpkg --listfiles  libcurl4-openssl-dev
/.
/usr
/usr/bin
/usr/bin/curl-config
/usr/include
/usr/include/x86_64-linux-gnu
/usr/include/x86_64-linux-gnu/curl
/usr/include/x86_64-linux-gnu/curl/curl.h
/usr/include/x86_64-linux-gnu/curl/curlver.h
/usr/include/x86_64-linux-gnu/curl/easy.h
/usr/include/x86_64-linux-gnu/curl/mprintf.h
/usr/include/x86_64-linux-gnu/curl/multi.h
/usr/include/x86_64-linux-gnu/curl/stdcheaders.h
/usr/include/x86_64-linux-gnu/curl/system.h
/usr/include/x86_64-linux-gnu/curl/typecheck-gcc.h
/usr/include/x86_64-linux-gnu/curl/urlapi.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libcurl.a
/usr/lib/x86_64-linux-gnu/libcurl.la
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libcurl.pc
/usr/share
/usr/share/aclocal
/usr/share/aclocal/libcurl.m4
/usr/share/doc
/usr/share/doc/libcurl4-openssl-dev
/usr/share/doc/libcurl4-openssl-dev/NEWS.Debian.gz
/usr/share/doc/libcurl4-openssl-dev/changelog.Debian.gz
/usr/share/doc/libcurl4-openssl-dev/changelog.gz
/usr/share/doc/libcurl4-openssl-dev/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/curl-config.1.gz
/usr/lib/x86_64-linux-gnu/libcurl.so
```

cefn, 2020-07-05:
The build process for 6.8.15 on the Kobo perhaps uses the target `kobo-libs` and not `libs` given https://github.com/XCSoar/XCSoar/blob/v6.8.15/Makefile#L120

So following these build steps

```
make TARGET=KOBO DEBUG=n USE_CCACHE=y kobo-libs
make TARGET=KOBO DEBUG=n USE_CCACHE=y
make TARGET=KOBO DEBUG=n USE_CCACHE=y output/KOBO/KoboRoot.tgz
```

...I have been able to progress to the second main build step which fails as follows with both tag v6.8.15 and branch v6.8.x...

```
In file included from src/Engine/Trace/Trace.cpp:24:
src/Engine/Trace/Trace.hpp:34:10: fatal error: boost/intrusive/list.hpp: No such file or directory
 #include <boost/intrusive/list.hpp>
```

Here is the full console output from `make TARGET=KOBO DEBUG=n USE_CCACHE=y` in the container...
 
```
root@0fbebc07bbcf:/opt/xcsoar# make TARGET=KOBO DEBUG=n USE_CCACHE=y          
  CXX     output/KOBO/opt/src/IO/ConfiguredFile.o
  CXX     output/KOBO/opt/src/IO/DataFile.o
  CXX     output/KOBO/opt/src/Airspace/ProtectedAirspaceWarningManager.o
  CXX     output/KOBO/opt/src/Airspace/ActivePredicate.o
  CXX     output/KOBO/opt/src/Task/Serialiser.o
  CXX     output/KOBO/opt/src/Task/Deserialiser.o
  CXX     output/KOBO/opt/src/Task/SaveFile.o
  CXX     output/KOBO/opt/src/Task/LoadFile.o
  CXX     output/KOBO/opt/src/Task/TaskFile.o
  CXX     output/KOBO/opt/src/Task/TaskFileXCSoar.o
  CXX     output/KOBO/opt/src/Task/TaskFileIGC.o
  CXX     output/KOBO/opt/src/Task/TaskFileSeeYou.o
  CXX     output/KOBO/opt/src/Task/DefaultTask.o
  CXX     output/KOBO/opt/src/Task/MapTaskManager.o
  CXX     output/KOBO/opt/src/Task/ProtectedTaskManager.o
  CXX     output/KOBO/opt/src/Task/FileProtectedTaskManager.o
  CXX     output/KOBO/opt/src/Task/RoutePlannerGlue.o
  CXX     output/KOBO/opt/src/Task/ProtectedRoutePlanner.o
  CXX     output/KOBO/opt/src/Task/TaskStore.o
  CXX     output/KOBO/opt/src/Task/TypeStrings.o
  CXX     output/KOBO/opt/src/Task/ValidationErrorStrings.o
  CXX     output/KOBO/opt/src/RadioFrequency.o
  CXX     output/KOBO/opt/src/Engine/Navigation/TraceHistory.o
  CXX     output/KOBO/opt/src/Engine/Navigation/Aircraft.o
  CXX     output/KOBO/opt/src/Engine/Trace/Point.o
  CXX     output/KOBO/opt/src/Engine/Trace/Trace.o
In file included from src/Engine/Trace/Trace.cpp:24:
src/Engine/Trace/Trace.hpp:34:10: fatal error: boost/intrusive/list.hpp: No such file or directory
 #include <boost/intrusive/list.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

cefn, 2020-07-05:
I have progressed beyond the boost error after manually running an extra apt install in the docker container's interactive shell, like `apt-get install libboost-dev`.

After this I arrive at a new error in both tag v6.8.15 and branch v6.8.x... 

```
  CXX     output/KOBO/opt/src/Event/Poll/Linux/Input.o
src/Event/Poll/Linux/Input.cpp: In member function 'void LinuxInputDevice::Read()':
src/Event/Poll/Linux/Input.cpp:133:9: error: 'errno' was not declared in this scope
     if (errno != EAGAIN && errno != EINTR)
         ^~~~~
src/Event/Poll/Linux/Input.cpp:133:9: note: 'errno' is defined in header '<cerrno>'; did you forget to '#include <cerrno>'?
src/Event/Poll/Linux/Input.cpp:35:1:
+#include <cerrno>
 #include <sys/ioctl.h>
src/Event/Poll/Linux/Input.cpp:133:9:
     if (errno != EAGAIN && errno != EINTR)
         ^~~~~
src/Event/Poll/Linux/Input.cpp:133:18: error: 'EAGAIN' was not declared in this scope
     if (errno != EAGAIN && errno != EINTR)
                  ^~~~~~
src/Event/Poll/Linux/Input.cpp:133:18: note: suggested alternative: 'KEY_AGAIN'
     if (errno != EAGAIN && errno != EINTR)
                  ^~~~~~
                  KEY_AGAIN
src/Event/Poll/Linux/Input.cpp:133:37: error: 'EINTR' was not declared in this scope
     if (errno != EAGAIN && errno != EINTR)
                                     ^~~~~
src/Event/Poll/Linux/Input.cpp:133:37: note: suggested alternative: 'VINTR'
     if (errno != EAGAIN && errno != EINTR)
                                     ^~~~~
                                     VINTR
make: *** [build/compile.mk:108: output/KOBO/opt/src/Event/Poll/Linux/Input.o] Error 1
```

cefn, 2020-07-05:
The `errno was not declared` error seems connected to the one reported in https://github.com/XCSoar/XCSoar/issues/197 and apparently fixed in https://github.com/XCSoar/XCSoar/commit/3ea0e52ae690cbd9b80812138a35f9f6b2d0678b but attempting to cherry pick this into the v6.8.15 tag or v6.8.x branch indicates that it has already been backported...

```
$ git cherry-pick 3ea0e52ae690cbd9b80812138a35f9f6b2d0678b
HEAD detached at v6.8.15
You are currently cherry-picking commit 3ea0e52ae6.

nothing to commit, working tree clean
The previous cherry-pick is now empty, possibly due to conflict resolution.
```

To get past this, I ended up repeating the change to `src/Event/Poll/Linux/Input.cpp` indicated by @nico-krais 

https://github.com/nico-krais/XCSoar/commit/d1789700cc5806d66b38ece6c67de171fd202326

This then finally got me to a completed build of a KoboRoot.tgz

closed 2020-07-05 by cefn

MaxKellermann, 2020-07-07:
If you still need to change the code, then the problem isn't solved.