XCSoar/XCSoar #450: Master branch fails to build libcurl for Android7

clydebarrow, 2020-07-08:
XCSoar versions having and not having the problem
-------------------------------------------------
7.0_preview15

Expected behavior
-----------------
App can be built for Android on MacOs using `make TARGET=ANDROID7`

Actual behavior
---------------
Building fails during the curl library build. 

Steps to reproduce the behavior
-------------------------------

Run `make TARGET=ANDROID7` with the current master HEAD on MacOs.

Do you have any idea what may have caused this?
-----------------------------------------------
The curl build fails as it does not set `HAVE_SOCKET`. This in turn is because the `configure` script checks for the presence of a prototype for `socket()` by running the preprocessor (CPP).

The Android build is performed with llvm, found in the Android NDK distribution. llvm implements CPP with `clang -E`, however `configure` runs this with arguments from `CPPFLAGS` which is different to `CFLAGS`. Since clang has to know the target architecture it requires a `-target` option, which is included in `CFLAGS` but currently not `CPPFLAGS`.  Without this the default include paths are not set up correctly, and in the test code `#include <sys/types.h>` fails.

Do you have an idea how to solve the issue?
-------------------------------------------

Add a `-target` option to CPPFLAGS.

PR to follow.

MaxKellermann, 2020-07-17:
As I wrote already in https://github.com/XCSoar/XCSoar/pull/451#issuecomment-655451814, this is a CURL bug. Your PR is wrong.

MaxKellermann, 2020-08-02:
Can't reproduce.
```
checking for select... yes
```

closed 2020-08-02 by MaxKellermann