XCSoar/XCSoar #454: Add Flat Triangle evaluation to XC calculations

lud-hu, 2020-07-14:
XCSoar version
--------------
current version

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Adding "Flat Triangle" as contest evaluation (in addition to "Classic" and "FAI Triangle").

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
I'm currently looking into using XCSoar's track log evaluation functionality (with the python wrapper). It is working very well to evaluate a GPS tracklog file for the XC distances (Classic, FAI Triangle).
I guess it's not a thing for actual gliders, but I want to evaluate paragliding tracklogs, where it is also interesting to have Flat Triangles evaluated.

> A Flat Triangle is a triangular flight that does not meet the strict requirements of an FAI Triangle, provided that no leg is shorter than 15% or longer than 45% of the total leg distance.
https://www.xcleague.com/xc/info/rulesflights-circuit.html#flat

I've found the place where the FAI Triangle rules are defined:
src/Engine/Task/Shapes/FAITriangleRules.cpp

Would it be possible to add the Flat Triangle as well? Unfortunately I'm not a C++ developer, but with some help and some ideas about where to start I maybe could add the Flat Triangle calculation and create a Pull Request for it.

Thanks!

lordfolken, 2020-07-14:
C++ isn't that hard. ;-)  Try and submit is the best advice i can give you.

MaxKellermann, 2020-07-15:
This part of the code is hard and complicated. There's so much simple frontend code which could and should be improved.
Much of the triangle calculation code is complicated, obscure and inconsistent. All of the rules are hard-coded. I've tried to improve the code yesterday to make it possible to implement new triangle rules, by moving the basic rules to a separate library.
Which is duplicate/redundant code, because there's `TriangleContest`/`OLCTriangleRules` for post-flight scoring, and there's `FAITriangleRules` for task planning. Those shall be merged into one unified ruleset, so one implementation can be used both for task planning and for scoring.
I'd welcome contributions to XCSoar very much, but this is not the best place to get started, as it will lead to frustration.

lordfolken, 2026-09-29:
Triage: flat-triangle scoring is a paragliding contest rule on top of the existing XC optimiser, which is already one of the harder parts of the engine. Niche benefit and a large change, so this goes to `backlog` instead of needs-triage.

Not closing. A PR is still the way in if you want it.